ArduPilot/ardupilot #24363: FBWA Not constraining throttle as prescribed

ChuckK4808, 2023-07-21:
## Bug report

**Issue details**

FBWA not constraining throttle to THR_MIN limits. During flight, FBWA Thr percentage dropped lower than THR_MIN causing an engine out scenario.

**Version**
v4.3.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Havoc Jet airframe with Kingtech K320G2 Turbine engine.

**Hardware type**
Cube Orange & Here 3 GPS

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2023-07-21:
We really need a log to look into this. Do you have `THR_PASS_STAB` set?

ChuckK4808, 2023-07-24:
*Logs Attached*

This log is a recreation of the problem I am encountering. When in FBWA and simulating an RC link loss (turning off the controller), the throttle drops to min, FS triggers and picks back up the throttle to meet the preprogrammed behavior for FS. The issue enlies on the drop of the throttle. I have thr_min set to 26% (percentage needed to keep the turbine running. The drop in the throttle will shutdown the turbine even though it picks the throttle back up. I changed flight mode to FBWB and this behavior does not happen. Any advice on why this behavior is happening would be greatly appreciated. 

Just to be clear, the parameters were copied from an actual aircraft and ran on a bench to see behavior. I use mavlink inspector on QGCS to see what the output was doing.
[FBWB.zip](https://github.com/ArduPilot/ardupilot/files/12141397/FBWB.zip)
[FBWA.zip](https://github.com/ArduPilot/ardupilot/files/12141411/FBWA.zip)

IamPete1, 2023-07-24:
@ChuckK4808 Thanks for the more detailed report. I have done a fix.  https://github.com/ArduPilot/ardupilot/pull/24385. This is a Plane 4.3.7 build for CubeOrangePlus. 
[CubeOrangePlus 4-3-7 throttle failsafe fix arduplane.zip](https://github.com/ArduPilot/ardupilot/files/12148378/CubeOrangePlus.4-3-7.throttle.failsafe.fix.arduplane.zip) Can you give it a try? Bench testing recommended before flight testing. Note it is also a few minor versions ahead of 4.3.3 from your log.

ChuckK4808, 2023-07-24:
@IamPete1 Would this fix also work for the Cube Orange?

ChuckK4808, 2023-07-25:
@IamPete1 I installed the firmware you sent me and I can confirm that it works. Would you like a copy of the logs? to confirm

IamPete1, 2023-07-25:
@ChuckK4808 that test log would be great, Thanks!

ChuckK4808, 2023-07-25:
@IamPete1 I will get that to you as soon as possible

PR #24385 (2023-07-26): Plane: fix throttle going bellow min in fbwa RC failsafe
fixes #24363 

This fixes a momentary bypassing of the set minimum throttle in FBWA/stabilize/acro/training/autotune. `has_valid_input` goes false for a few seconds before the failsafe action kicks in and changes modes. This can lead an ICE engine to stop.

[image]

Our throttle handling is very complex, it is next getting near to the top of my list for re-work.
closed 2023-07-26 by tridge