ArduPilot/ardupilot #24402: CAN2 - Failed to register CANSensor

Sekilsgs2, 2023-07-26:
When using can2 for benewake protocol (or USD1) - gives a sensor registration error - 
```
26.07.2023 22:21:58 : Failed to register CANSensor Benewake
```

Cuav X7 pro
Arducopter 4.3.5 or master
[image]

xianglunkai, 2023-07-27:
@Sekilsgs2 
Perhaps you can connect a console to see the current working status or errors of all CANs

xianglunkai, 2023-07-27:
Or view relevant records in the corresponding log

xianglunkai, 2023-07-27:
One possible reason is that you may only have one CAN when building the system. You can use candump can1 to view the current status, if your system is Linux

Sekilsgs2, 2023-07-27:
> @Sekilsgs2 Perhaps you can connect a console to see the current working status or errors of all CANs

DroneCAN working good for can2 but rangefinders with can protocol benewake or usd1 -not working.

Sekilsgs2, 2023-07-29:
https://github.com/ArduPilot/ardupilot/issues/24415

closed 2023-07-29 by Sekilsgs2