ArduPilot/ardupilot #24423: IMU Data Loss in Copter 4.3.7 Stable (mRoControlZeroH7)

joshanne, 2023-07-28:
## Bug report

**Issue details**

**Version**
Copter-4.3.7

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware Configuration**
* mRoControlZeroH7 (Copter-4.3.7 - latest stable release)
* mRo SAM GPS + IST8308 Mag (Full Size) M10037 (See: https://store.mrobotics.io/product-p/m10037a.htm)

**Description**

We are experiencing data loss for the IMUs when the Flight Controller is configured with External GPS/Compass module. When the module is not present, there is no data loss in the IMUs.

We can replicate (and validate the proposed solution) with the following parameter set.

```
EK3_IMU_MASK  7
LOG_REPLAY 1
```

1. Enable `LOG_DISARMED`
2. Reboot
3. Generate ~1min log data
4. Disable `LOG_DISARMED`
5. Reboot
6. Download log and check with `graph sum(RISI[0].DADT,1) sum(RISI[1].DADT,2) sum(RISI[2].DADT,3) IMU[0].TimeUS/1E6`

Looking at the log, we observe a time slip in the recorded data.

[image]

**Solution**

It appears for this flight controller the I2C DMA is interfering with the IMUs, resulting in data loss.

Cherry-picking these upstream commits (on top of Copter-4.3.7) addresses the issue seen. https://github.com/ArduPilot/ardupilot/pull/23568/commits

With these changes, we can see all three IMUs are being read without data loss, and the accumulated time rises at the same gradient as elapsed time.

[image]

Two log files are attached in this zip. They contain:
1. The issue present, on Copter-4.3.7 stable (_log-copter-4.3.7-issue-present.bin_)
2. The issue resolved, Copter-4.3.7 with cherry-picked commit (log-copter-4.3.7-fixed-with-cherrypick-commits)

[log-copter-4.3.7.zip](https://github.com/ArduPilot/ardupilot/files/12190867/log-copter-4.3.7.zip)

joshanne, 2023-07-31:
@rmackay9 Just checked and this is appears to still be an issue in Copter 4.4 beta3 (latest available beta for Copter).

andyp1per, 2023-07-31:
Can you try this change with 4.4? https://github.com/ArduPilot/ardupilot/pull/24454 - I think the problem is the DMA sharing

tridge, 2023-07-31:
@andyp1per I think this alternative should be better:
https://github.com/ArduPilot/ardupilot/pull/24456

joshanne, 2023-09-04:
Given this was picked up in Copter-4.4.0, happy to close this off.

closed 2023-09-04 by joshanne