ArduPilot/ardupilot #24444: Bug report

tanz-nwpu, 2023-07-30:
when i run "sim_vehicle.py -f quadplane --console --map"，the error is as follows:

tan@tz:~/ardupilot/ArduPlane$ sim_vehicle.py -f quadplane --console --map
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/tan/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/tan/ardupilot 
Setting out to                           : /home/tan/ardupilot/build 
Autoconfiguration                        : enabled 
Checking for program 'python'            : /usr/bin/python3 
Checking for python version >= 3.6.9     : 3.6.9 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 7.5.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Enabling -Werror                         : yes 
Enabled OpenDroneID                      : no 
Enabled firmware ID checking             : no 
GPS Debug Logging                        : no 
Enabled custom controller                : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
DC_DSDL compiler                               : /home/tan/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : maybe 
Scripting runtime checks                       : enabled 
Debug build                                    : disabled 
Coverage build                                 : disabled 
Force 32-bit build                             : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (0.720s)
{'waf_target': 'bin/arduplane', 'default_params_filename': 'default_params/quadplane.parm', 'model': 'quadplane', 'sitl-port': True}
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/tan/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arduplane"
Waf: Entering directory `/home/tan/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Embedding file models/plane-3d.parm:Tools/autotest/models/plane-3d.parm
Embedding file models/plane.parm:Tools/autotest/models/plane.parm
Embedding file models/xplane_heli.json:Tools/autotest/models/xplane_heli.json
Embedding file models/xplane_plane.json:Tools/autotest/models/xplane_plane.json
[ 332/1145] Compiling libraries/AP_Mount/AP_Mount_Viewpro.cpp
[ 393/1145] Compiling libraries/AP_OLC/AP_OLC.cpp
[ 394/1145] Compiling libraries/AP_OSD/AP_OSD_Setting.cpp
[ 395/1145] Compiling libraries/AP_OSD/AP_OSD_SITL.cpp
[ 396/1145] Compiling libraries/AP_OSD/AP_OSD_MSP.cpp
[ 397/1145] Compiling libraries/AP_OSD/AP_OSD.cpp
[ 398/1145] Compiling libraries/AP_OSD/AP_OSD_MSP_DisplayPort.cpp
[ 399/1145] Compiling libraries/AP_OSD/AP_OSD_Backend.cpp
[ 400/1145] Compiling libraries/AP_OSD/AP_OSD_MAX7456.cpp
[ 401/1145] Compiling libraries/AP_OpenDroneID/AP_OpenDroneID_DroneCAN.cpp
[ 402/1145] Compiling libraries/AP_OpenDroneID/AP_OpenDroneID.cpp
[ 403/1145] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_MAV.cpp
[ 404/1145] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_Calibrator.cpp
In file included from ../../libraries/AP_Mount/AP_Mount_Viewpro.cpp:1:0:
../../libraries/AP_Mount/AP_Mount_Viewpro.h:262:37: warning: ‘AP_Mount_Viewpro::E1Packet::<unnamed struct>::source’ is too small to hold all values of ‘enum class AP_Mount_Viewpro::TrackingSource’
             TrackingSource source : 3;  // see TrackingSource enum above
                                     ^
../../libraries/AP_Mount/AP_Mount_Viewpro.cpp: In member function ‘bool AP_Mount_Viewpro::send_tracking_command(AP_Mount_Viewpro::TrackingCommand, uint8_t)’:
../../libraries/AP_Mount/AP_Mount_Viewpro.cpp:637:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/home/tan/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_Mount' failed (exit status 1): 
	{task 140093594717648: cxx AP_Mount_Viewpro.cpp -> AP_Mount_Viewpro.cpp.0.o}
 (run with -v to display more information)
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks

IamPete1, 2023-07-30:
The error is resolved in newer compiler versions. We make compiler a version a hard requirement for ChibiOS builds, but not SITL. I guess we might need to add a minimum version check for SITL too.

rmackay9, 2023-08-01:
Thanks for the report.  We're going to do a couple of things:

1. find a work around to make the older compiler happy
2. add a compiler minimum version check to give developers a better idea of what is actually wrong

tanz-nwpu, 2023-08-02:
> 

Thank you very much for your solution, I just tried it and it compiled successfully.

closed 2023-08-02 by rmackay9