ArduPilot/ardupilot #24445: EKF failsafe untriggered on high errors, resulting in a crash #10

SudharssanMohan, 2023-07-30:
### Issue details

Reporting one of the discovered instance of an ArduCopter SITL execution , where the Extended Kalman Filter does not trigger fail safe landing, despite significant errors in Copter's mission resulting from compromised accelerometer values. EKF undergoes multiple lane and core switches, drone crashes before completing assigned mission.

Significant deviation in X velocity in the mission. (Other controllers have errors too, this is an sample to show the impact, logs are attached to aid in deeper analysis)
[image]

### Version
ArduCopter V4.5.0-dev 

### Platform
[X] Copter
[X] Ardupilot SITL simulation

### Logs
Attached is the telemetry logs of the mission,BIN file for further log analysis and the mission used in this execution.
[logs.zip](https://github.com/ArduPilot/ardupilot/files/12210260/logs.zip)

Mission plan used:

[mission.txt](https://github.com/ArduPilot/ardupilot/files/12210259/07-23-04-27-59.txt)

### Context
As part of my research in contributing towards making Ardupilot more resilient to sensor injection attacks , I have modified AP to inject malicious yet realistic(can be generated in the real world) accelerometer values and run different missions. Some executions (such as the one described in this report , successfully crashes the drone without triggering the EKF3's failsafe emergency landing, which could lead to physically damaging the drone instead of minimizing the damage in a real attack scenario)

IamPete1, 2023-07-31:
Could these be combined into a single summary issue? I don't think having 10 similar issues really brings any value, these are now taking up over half of the first page of issues meaning we might miss more urgent problems.

SudharssanMohan, 2023-08-01:
@IamPete1 @rmackay9 Thank you for your patience and sorry about that (should I close my other issues ?), I have combined all my reports into this one single report. I am awaiting your response to see how this goes. As @rmackay9 previously commented in one of my initial report, it seems like EKF3 is unable to handle situations like the ones I reported resulting in direct crashes without any failsafe land to mitigate damages.

I am hoping to get some acknowledgement (maybe an explanation on this behavior down the line?) on this, as it is an active work in progress on my side it would help me further if I get some feedback.

SudharssanMohan, 2023-08-23:
Hello @IamPete1 @rmackay9 . I just have a follow up question on this issue as I am actively working on it.
The following is the EKF3 velocity innovations for N,E and D axis for case no.7. To my understanding, with high innovations and ample position and velocity errors, the fail safe should have triggered to minimize any damages to the drone. But no such thing happens and drone deviates way off path before it ultimately crashes.

[image]
[image]

This is just one case that I am pointing out as an example. But this question stands in all the above cases as to why EKF3 did not catch such significant errors in the system. 
Is it because EKF3 is not capable of detecting such errors when calibrated for normal flight as @rmackay9 previously pointed out?