ArduPilot/ardupilot #24482: The problem when I use the SITL with Gazebo

tanz-nwpu, 2023-08-02:
Using SITL with Gazebo in Ubuntu 18.04, I met the problem as follow:
[image]
I set the ASRPD_FWB_MIN as 14.0m/s, Q_ASSIST_SPEED 12.0m/s, but when I run the STIL with Gazebo, the Airspeed is 12m/s, which is lower than  ASRPD_FWB_MIN , and the motor of the rotor is engaged in operation. Is this imagination normal?

Ryanf55, 2023-08-07:
Have you followed this guide? 
https://ardupilot.org/plane/docs/tecs-total-energy-control-system-for-speed-height-tuning-guide.html#set-initial-parameters-throttle-pitch-airspeed-and-vertical-speed-limits

I would watch the following section of @priseborough's video: https://youtu.be/kHFqUgTwT5c?t=1238

If that doesn't help, please attach your param file to this ticket.

peterbarker, 2025-08-12:
Closing this one as no response.

closed 2025-08-12 by peterbarker