ArduPilot/ardupilot #24493: The ServoX_Function: 81 is not available for Quadplane

dhruvilodonata, 2023-08-03:
## Bug report
The ServoX_Function: 81 is not available for Quadplane

**Issue details**
Q_M_BOOST_SCALE (0 to 5) is available for the booster motors, while to assign the servo function; BoostThrottle (81) is not available for any servo.

The ServoX_Function: 81 (BoostThrottle) is available in Copter firmware.

_Please describe the problem_

**Version**
Arduplane 4.3.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
4 Tilt Rotor in H frame plus 2 booster motors near fuselage

**Hardware type**
Pixhawk 6C

IamPete1, 2023-08-03:
Should work fine if you set it manually in the full parameter list.

dhruvilodonata, 2023-08-04:
I tried.

The Servo 9 to 12 are motor 1 to 4 (tilt, Dshot) and I assigned servo 13 and 14 as 81 (boostthrottle) from full parameter list, but when i tried there was no response from those two booster motors. The green bar for servo 13 and 14 in setup -> mandatory HW -> servo was also blank.  I armed and gave full throttle to check whether at full throttle any response is there for booster motors, but there was none.

IamPete1, 2023-08-04:
Do you have a log showing it not working?

dhruvilodonata, 2023-08-05:
I don't have log at the moment but on Monday I can provide you. It was just a ground test (bench test).

dhruvilodonata, 2023-08-07:
Hello, I tried today again with the same parameters on mission planner with Qstabilized mode and physical connections like the last week and it worked at somewhere around 50% of throttle with Q_M_BOOST_SCALE as 0.2. 

Thanks for the prompt support.

closed 2023-08-12 by auturgy