ArduPilot/ardupilot #24521: Provide VSCode launch and debug tasks for developers to build and debug SITL

Ryanf55, 2023-08-06:
## Feature request

**Is your feature request related to a problem? Please describe.**

Due to ArduPilot's highly custom simulation bringup (sim_vehicle.py), and lack of use of CMake or other common build tool, it's difficult to use VSCode's GUI support for configuring, building, testing, and debugging. 

Developers are maintaining [their own custom configs](https://github.com/ArduPilot/ardupilot_vscode_devenv/issues/1#issuecomment-1214816404), but these aren't shared.

**Describe the solution you'd like**

Merge one of the dev's `launch.json` and `tasks.json` files that works on `master`, and then allow others to start contributing. When working, this provides the ability to use a keyboard shortcut to configure, build, run `sim_vehicle.py`, attach `gdb`, and inspect the code at a breakpoint WITHOUT using `gdb's` CLI. This can be faster and easier, but it still allows console-commands for gdb.

Here is a hard-coded launch and task showing a breakpoint being triggered in SITL. 
[image]

**Describe alternatives you've considered**

https://github.com/ArduPilot/ardupilot_vscode_devenv/issues/1

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

TompsonTan, 2023-08-10:
I think it's necessary to add more support to use VSCode， we need an efficient way for code reading,editing, and debug.

bugobliterator, 2025-05-10:
This feature is now available through current pre-release of ardupilot vscode extension. So probably this issue can be closed now @Ryanf55

peterbarker, 2025-08-12:
Closing this one now - Sid's got it working nicely!