ArduPilot/ardupilot #24653: TECS speedweight scaling (when TECS_LAND_SPDWGT is -1) is broken

rubenp02, 2023-08-15:
## Bug report

**Issue details**

_Please describe the problem_

**Version**
Latest dev (4.5)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Any

**Hardware type**
Any

The speedweight scaling is broken, works in latest stable release.
[logs.zip](https://github.com/ArduPilot/ardupilot/files/12344162/logs.zip)
In 4.3.7.BIN, graphing TECS.w reveals the expected behaviour. In 4.5.BIN, graphing TEC2.PEW shows that the speedweight doesn't get scaled during landing at all.

IamPete1, 2023-08-15:
Looks like there both custom branches?

rubenp02, 2023-08-15:
> Looks like there both custom branches?

They're my fork of the master and Plane-4.3 branches but they have no changes. You'll get the same behaviour by cloning the main repo's branches.