ArduPilot/ardupilot #24655: Plane: Allow nudging of horizon level via transmitter when in non-altitude-hold modes

mikedkelly, 2023-08-15:
The docs describe FBWA, Training, and Stability flight modes as effective ways to maiden new models and for beginners to learn to control an aircraft before transitioning to manual mode. However, if the horizon level calibration is incorrect, the pilot has no way to "trim" the model in flight. He is left with holding a corrective demand throughout the model's flight in order to maintain a level attitude. This is very distracting and undesirable during a maiden or as a beginner pilot. Aborting the flight and recalibrating level attitude on the ground is not only cumbersome, but without the use of precise tools, will rarely result in an accurate pitch trim for the model.

Since the use of transmitter trim causes adverse effects in these flight modes, I propose something like a new tuning parameter in transmitter based tuning where a slider or pot on the transmitter could be set to nudge either AHRS_TRIM_Y or TRIM_PITCH_CD (and the equivalent roll settings). The reset and save functionality of transmitter based tuning should apply as well. This allows the pilot to correct their setup while continuing to fly as well as make additional adjustments resulting from throttle changes if the mode's attitude hold is not adequate. 

I've considered methods of iteratively correcting the mechanical linkages as documented, as well as simply using other modes such as FBWB and Acro mode, but it still comes back to having a properly calibrated level setting, and currently that can only be adjusted on the ground. While making changes to the level calibration in the air can have adverse consequences if not used correctly, I feel the risk is no greater than setting it incorrectly on the ground and taking flight.

While primarily useful for LOS fixed wing flying, I can also see this as a useful enhancement to any model type.

IamPete1, 2023-08-15:
We do have a "save trim" RC option on copter, it just sets the current attitude to be 0. We could implement it on plane I suppose.

However I don't think the "make it easier for beginners" argument is really a valid one. We should now say that they need to setup this extra channel as well as there flight modes and control inputs? And that they should then twiddle the new channel as there flying for the first time? Much better to just have clear instructions on how to get the trim correct in the first place.

mikedkelly, 2023-08-15:
I think that would be a good solution. It allows you to use the transmitter trims in flight, which is the procedure pilots are used to, but gives you a layer of safety by only making the adjustment to the level calibration once you have landed. I was always under the impression that you should never adjust transmitter trim in a stabilized mode, but if this works I am all for it.

peterbarker, 2025-08-12:
> I think that would be a good solution. It allows you to use the transmitter trims in flight, which is the procedure pilots are used to, but gives you a layer of safety by only making the adjustment [...]

We can't work backwards from RC trims to level calibrations - there's no way to map that.

@IamPete1 's "get the trim right in the first place" is probably the best advice.

Having said that, we could potentially make Copter's "RC Channel tuning" available on all vehicles.  This would allow you to directly modify the AHRS trim parameters on the transmitter - either on pots or via some *very* fancy stuff using the trim tabs and custom mixing.

mikedkelly, 2025-08-12:
The copter solution is a step in the right direction, but it doesn't really solve the underlying problem. Aerodynamically, an aircraft with positive pitch stability will change its level pitch trim with every change in thrust. So in pilot-controlled modes like FBWA, Training, and Stabilize, we need to be able to adjust trim on the transmitter during flight as we change throttle settings without it causing adverse effects to the autopilot. I think you answered @IamPete1 's advice though with your statement above it - you can't "get the trim right in the first place" if there is no way to set the correct level calibration based on a trim value - manually or otherwise. You can't just look at an aircraft on the ground and be able to know intuitively or measure what the trim setting should be.

mikedkelly, 2025-08-14:
That is a good point. I guess level calibration is more of a gyro angle vs. pitch trim being a climb rate. Perhaps as IamPete1 said above, a good instruction doc is the best solution. Something like: manually trim your plane to level in FWBA at your desired cruise throttle, then use the logs to get the angle of attack during level flight. Use some formula f(x) to convert that to the level calibration angle and set that parameter. Then reset your trims back to 0 and fly again. Decide on how you want the plane to behave with changes to throttle and adjust the pitch trim vs. throttle parameter. I am just spoiled with modern programmable transmitters that let you create multi-point mix curves that you can dynamically modify in-flight with a pot on your transmitter.