ArduPilot/ardupilot #24665: Regression in AP_ESC_Telem: Fix some time wrap issues that can lead to using stale data if a ESC stops responding

casrya, 2023-08-16:
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

Regression in commit 935fad54adaa48a51ba572cf4995d770e481cf4d AP_ESC_Telem: Fix some time wrap issues that can lead to using stale data if a ESC stops responding

There appears to be a bug on line 111 in the method AP_ESC_Telem::get_active_esc_mask() in libraries/AP_ESC_Telem/AP_ESC_Telem.cpp.

I believe it should be “&& !rpm_data_within_timeout(…)” to match the behaviour prior to this commit which in my case was working but no longer does (without the fix of adding ‘!’).

As it is now it means that if you don't get non-RPM telem AND you did get RPM telem then continue (i.e. discard the data) meaning that unless you are also using EDT or have some other telem sensors (apart from basic BDSHOT RPM) you will not get any BDSHOT RPM data.

I think this was probably just a mistake with the refactor to the rpm_data_within_timeout() method in this commit but I am happy to leave it to the original author @WickedShell to confirm. That said I can advise that changing the condition as described above worked for me on a Matek H743 with a T-Motor F45A V2 4in1 running the latest BLHeli32 release.

Cheers,

casrya

**Version**
latest master at specified commit.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad-X

**Hardware type**
Matek H743 with a T-Motor F45A V2 4in1 running the latest BLHeli32 release

**Logs**
NOTED in OSD display while props spinning

WickedShell, 2023-08-16:
@casrya You're absolutely correct, that was a bad regression, and the system I was using in testing had enough data that I wasn't seeing the bug. #24672 should address it. I'm sorry about that.

amilcarlucas, 2023-08-19:
Closing this one, please reopen if it presists.

closed 2023-08-19 by amilcarlucas