ArduPilot/ardupilot #24680: FTP download log causing abnormal EKF posture settlement

lixiaowei123, 2023-08-17:
I recently used the wireless remote control app to download logs, resulting in abnormal EKF posture settlement. The transmission baud rate is 921600, and the firmware version is Copter-4.0.7; Uncontrollable unlocking attitude

andyp1per, 2023-08-17:
This is to be expected. Log download pegs the CPU which means the EKF gets very upset.

lixiaowei123, 2023-08-17:
@andyp1per Is all the code like this

lixiaowei123, 2023-08-18:
@timtuxworth Downloading logs should not affect others

lixiaowei123, 2023-08-18:
@andyp1per This is abnormal, programming bug, please do not close my

andyp1per, 2023-08-18:
H7 has more CPU to spare so the EKF errors are less but will occur if you choose a big enough file. If you don't want this to happen then you need to sacrifice download performance which no-one wants as download speed is already slow enough.