ArduPilot/ardupilot #24686: download log causing abnormal EKFor Gps Run

lixiaowei123, 2023-08-18:
I use the F4 microcontroller to sample the APP download logs on the wireless remote control. After the download is completed, the posture will be abnormal。BUG

amilcarlucas, 2023-08-19:
Please observe this:
1. never download logs while the vehicle is flying.
2. After the download is completed, you must reset the flight controller before arming the vehicle

rmackay9, 2023-08-20:
@lixiaowei123,

I don't think this qualifies as a bug but in any case, I've changed the title and made it clear that it is an enhancement request.

BTW, I don't think it's possible to download logs while the vehicle is flying and we have various arming checks that, if left enabled, should stop the vehicle from arming until the EKF has regained a good estimate.  If you find that it is possible to arm with a bad EKF then we could look at improving the arming checks.

Please understand though that may be difficult to both download logs at maximum speed and maintain a good EKF estimate.  There may be no way to have both meaning that we end up with user complaints on both sides.. some because the EKF estimate isn't good, from others because the log download is too slow.

lixiaowei123, 2023-08-21:
Thank you very much. My operations are all done while locked. Currently, I have added that whenever downloading logs or refreshing the log list, the flight control program must be restarted. I believe this is safe. Unlocking download logs is a mandatory limitation. Currently, STM32H7 has almost no issues, and F4 appears almost every time.