ArduPilot/ardupilot #24688: motor4 output is 0

guogeWS, 2023-08-18:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

In helicopter firmware, when H_TAIL_TYPE is set to CW or CCW mode, the motor4 output is 0, the motor device cannot be initialized, and the motor4 output is only available after unlocking

**Version**
4.5.0_dev

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
helicopter

**Hardware type**
pixhawk 6x

**Logs**

IamPete1, 2023-08-19:
Do you have a log that shows this? What do you mean it works after unlocking?

guogeWS, 2023-09-04:
I mean, after the flight control is started, motor4 should output the minimum value of this channel, such as 1100, but it is 0, and the value will be set only after the aircraft is armed