ArduPilot/ardupilot #24691: AP_MOUNT: Missing sanity check for MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW mission item

trolledbypro, 2023-08-18:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

When testing MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW mission items that contain "NaN" params, the param is rejected and a MAV_MISSION_INVALID_PARAMx is returned.

Wireshark capture of the MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW command:
[image]
Wireshark capture of the MISSION_ACK reply:
[image]

The code handling DO_GIMBAL_MANAGE_PITCHYAW in AP_MOUNT contains a sanity check that allows "NaN" params when either both pitch/yaw angle or both pitch/yaw turn rates are "NaN".

**Version**
ArduPlane SITL built from the main branch (current as of 18 August 2023)
**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
SITL

**Logs**
MAVSDK program log:
```[01:21:42|Info ] MAVSDK version: v1.4.0-514-g9e06fc48 (mavsdk_impl.cpp:23)
[01:21:42|Info ] New system on: 172.29.176.1:51990 (with system ID: 1) (udp_connection.cpp:192)
[01:21:42|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:694)
[01:21:42|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[01:21:42|Warn ] Vehicle type changed (new type: 1, old type: 0) (system_impl.cpp:222)
[01:21:42|Debug] Discovered 1 component(s) (system_impl.cpp:548)
Waiting for system to be ready
[01:21:42|Debug] Using gimbal protocol v2 (mission_impl.cpp:107)
[01:21:42|Debug] Using Gimbal Version 2 as gimbal manager information for gimbal device 1 was discovered (gimbal_impl.cpp:71)
System ready
Creating and uploading mission
Uploading mission...
Mission upload failed: Invalid Argument, exiting.```

trolledbypro, 2023-08-18:
This is something I can fix quickly, I can replicate the sanity check from AP_MOUNT and PR.

rmackay9, 2023-08-18:
I'm not sure that AP supports NaNs in mission command arguments.

If you think you've got a possible fix then that would be great, thanks!

trolledbypro, 2023-08-18:
I have this implemented in my branch. The mission code for MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW does support NaN as arguments: https://github.com/ArduPilot/ardupilot/blob/7d264f1700d1d931ec401774be71bc170f158157/libraries/AP_Mission/AP_Mission_Commands.cpp#L288C2-L300