ArduPilot/ardupilot #24703: HDZero MSP joystick commands from pitch axis are inverted

aviphysics, 2023-08-18:
## Bug report

**Issue details**

HDZero MSP joystick commands from pitch axis (right Y stick in mode 2) are inverted.

For example, 
* commanding pitch up will move down in the menu and vice versa
* to enter VTX menu on a mode 2 transmitter, one most move the left stick down+in and right stick+in. The command should be both sticks down and in.

**Version**
Arducopter V4.4.0-beta 1

**Platform**
[maybe] All (it may be in all versions, but I only have arducopter)
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Kakute H7 mini

**Logs**
If there are any helpful logs, please let me know. It seems pretty straight forward.

rmackay9, 2023-08-18:
Maybe @andyp1per can chime in on this one..

andyp1per, 2023-08-19:
I'm not sure I understand how the MSP menu commands work, do you have a CAM setup somehow? Please can you post your parameters

aviphysics, 2023-08-19:
> I'm not sure I understand how the MSP menu commands work, do you have a CAM setup somehow? Please can you post your parameters

I will post parameters later, but the VTX is connected to the flight controller over MSP on a uart port. Somehow it gets joystick movements and that allows you to enter commands to change VTX and camera settings.

aviphysics, 2023-08-19:
[
[image]

Here are how the commands should work, from the HDZero VTX manual. They still work, but you have to invert the vertical axis on the right stick.

aviphysics, 2023-08-19:
> I'm not sure I understand how the MSP menu commands work, do you have a CAM setup somehow? Please can you post your parameters

Param file attached
[AOS35 build 230819.param.txt](https://github.com/ArduPilot/ardupilot/files/12386039/AOS35.build.230819.param.txt)

andyp1per, 2023-08-19:
ArduPilot is doing nothing to provide this input. It's all down to how MSP displayport is interpreting the sensor data. There is no "fix" here, I suspect its just that usually pitch is inverted on ArduPilot - if you uninvert pitch and then invert it on your transmitter I suspect it will work.

aviphysics, 2023-08-19:
> ArduPilot is doing nothing to provide this input. It's all down to how MSP displayport is interpreting the sensor data. There is no "fix" here, I suspect its just that usually pitch is inverted on ArduPilot - if you uninvert pitch and then invert it on your transmitter I suspect it will work.

If it is forwarding the raw values, why does iNav and BF behave according to the manual?

aviphysics, 2023-08-19:
Oh dammit... Ardu must be using inverted pitch or something. Looks like when I setup the transmitter, I had inverted the pitch on the transmitter to compensate for Ardupilot.

I guess I need to figure out how to univert pitch control on AP

peterbarker, 2025-08-12:
> Why would the default Arducopter control have stick forward pitch backwards?

Poor life choices.

And an unwillingness to mass-break everybody's setups to fix it :-)

Closing this one now as I think it's adequately explained.