ArduPilot/ardupilot #24721: SITL Environment Problem

jankrystianserafim, 2023-08-21:
## Bug report

SITL Problem

I am using Ubuntu on my Windows laptop and trying to run the following command: ```
cd ~/ardupilot/ArduPlane
../Tools/autotest/sim_vehicle.py --console --map
```
I am getting the following error message:
<img width="834" alt="image" src="https://github.com/ArduPilot/ardupilot/assets/56222672/e71f2628-f223-465c-b6c4-1059a1ab05c4">

I followed the instructions [here](https://ardupilot.org/dev/docs/building-setup-windows10.html#building-setup-windows10) and it is not working. Any help would be appreciated, as SITL is causing me problems.

My goal is to be able to have a simulated SITL quadplane that I can connect a custom app that's running on a virtual Android device to. Would using MP SITL work, or do I have to continue trying this?

jankrystianserafim, 2023-08-21:
I am getting this problem too:

[image]

be-ska, 2023-10-06:
You linked the guide for WSL, but you're using Ubuntu. Have you followed this instructions?
1. https://ardupilot.org/dev/docs/building-setup-linux.html
2. https://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html

It should works straightforward on Ubuntu

peterbarker, 2024-05-07:
@jankrystianserafim did you resolve your issue here?  I can't reproduce here (the odd thing in the above is the lack of window system, but things just generally work here as I've tested them)

peterbarker, 2024-07-31:
OK, closing this as no response.  I hope the author got things worked out!

closed 2024-07-31 by peterbarker