ArduPilot/ardupilot #24729: Rover QuickTune: Excessive setting parameters causing oscillation

xianglunkai, 2023-08-22:
I found that PSC_ The parameters set by VEL are too large, resulting in oscillation behavior after turning at large angles. 
My personal opinion is:

1. Set P first instead of D, and D should be relatively small to prevent noise diffusion;

2. I can be set to 0 or 10% of the P value;

3. The maximum setting of D value should be 20% of P value;
4.
[image]

[image]

rmackay9, 2023-08-22:
Thanks for this.  I have a few ideas:

1. remove the velocity tune because it doesn't work well enough
2. the Slew Rate calculator can only detect relatively high frequency oscillations (3hz or higher).  Rover/Boat oscillations may be lower frequency than this meaning that we need to increase the SlewLimiter.cpp's WINDOW_MS definition to 1000 (or higher).  In my simulator tests however this did not help
3. try a different approach to find a good tune.  For example we could set the I gain to zero, then increase P until the root-mean-square error reduces.. perhaps then we could continue increasing it until it starts to deteriorate.

xianglunkai, 2023-08-22:
@rmackay9 
I agree with your suggestion 1. In fact, manual adjustment of PSC is not difficult and the effect is acceptable. 
Also, why do we always start setting from D instead of P?

rmackay9, 2023-08-22:
@xianglunkai,

Cool, I'll go ahead and remove it then (or you can of course).

I think we do D before P because tuning P may lead to us needing to reduce D.

rmackay9, 2023-11-25:
I've created a PR here to greatly simplify the rover quick tune https://github.com/ArduPilot/ardupilot/pull/25629

rmackay9, 2024-03-25:
We've removed the velocity tuning from the quick tune process so I'll close this issue.

closed 2024-03-25 by rmackay9