ArduPilot/ardupilot #24730: DISARM_CHECK bitmask for each disarm method

hinkz, 2023-08-22:
## Feature request

**Is your feature request related to a problem? Please describe.**

External software cannot force disarm via mavlink even when desired past version 4.1.

**Describe the solution you'd like**

ARMING_CHECK has a bitmask for the checks prior to arming. The same idea could be applied to a  DISARM_CHECK bitmask

DISARM_CHECK would be a configurable bitmask with parameters being
mavlink_disarm_check_is_flying 
rudder_disarm_check_is_flying - this is a strange option, and  we have not use for this but someone may?
more parameters may be added later

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X] Plane
[ ?] Rover
[ ? ] Submarine

rmackay9, 2023-08-22:
Hi @hinkz,

You can actually force disarm the vehicle using mavlink.  The magic number is listed on the [developer wiki here](https://ardupilot.org/dev/docs/mavlink-arming-and-disarming.html).

hinkz, 2023-08-23:
Thank you for your reply.

Should this also work if arming_require = 0 since 4.2? 
[image]

I'm going to test later today.

rmackay9, 2023-08-23:
@hinkz,

I suspect that if arming-require=0 then it will not be possible to disarm the vehicle but I'm not very familiar with this feature because we don't have it on Copter.  Rover has it but I'm not a fan of the feature.

hinkz, 2023-08-23:
In most use cases this would undesirable and potentially dangerous but having the configurable ability to do this while arming_require=0 fits my use case very well.

peterbarker, 2024-11-23:
I'm not really understanding this.  ARMING_REQUIRE=0 means you don't require the vehicle to be armed to operate.  It's *not* always-armed - it's just arming-isn't-required.  disarming the vehicle here doesn't make sense - it isn't actually ever armed!

See https://github.com/ArduPilot/ardupilot/issues/28644 - a proposed arm-once-when-ready feature.

peterbarker, 2025-04-29:
@hinkz any more thoughts?

hinkz, 2025-04-29:
Sorry its been a while. This request stemmed from a use case where ardupilot was being used as an external sensor rather than an autopilot. The EKF output differs depending on arm state and for this reason we operated in a forced armed state but have the need for the occasional disarm (e.g. so we can perform a mag cal) which would have been desirable over mavlink.

peterbarker, 2025-05-01:
> Sorry its been a while. This request stemmed from a use case where ardupilot was being used as an external sensor rather than an autopilot. The EKF output differs depending on arm state and for this [...]

OK, so since you looked we have sprouted a special "black box logging" mode which would probably be ideal for you.

We're also (probably!) going to merge a patch into Rover that arms the vehicle once the vehicle is passing prearm checks, as an alternative to requires_arming=0.  Combined with a GPIO output to indicate arm status this could be reasonably interesting as a logging solution.

I'm going to close this one now as the "REQUIRES_ARMING" thing really isn't a matter of being armed or disarmed, it's just not-requiring-arming, thus "disarm" isn't really a thing.

closed 2025-05-01 by peterbarker