ArduPilot/ardupilot #24760: Auto Mission fail with RP Lidar A2 integrated

Prasanna-icefire, 2023-08-25:
## Bug report
Auto Mission fail with RP Lidar A2 integrated

**Issue details**
With the obstacle avoidance parameters enabled and the RPLidar A2 integrated, the drone fails to execute the auto mission. The drone is observed to go in circles around the first way point and never reached the next waypoint.
In loiter mode the obstacle avoidance with RPLidar A2 works without issues.

**Version**
Firmware version 4.2.x

**Platform**
Only tried it on Copter

**Airframe type**
Air Frame - quad, V tail

**Hardware type**
Cube Orange

rmackay9, 2023-08-25:
@Prasanna-icefire,

Thanks for the report.  In general it's best to provide a log file or it's really all just guess work.  Also it is best to report issues first in the forums so for Copter-4.4.0 this is the place https://discuss.ardupilot.org/c/arducopter/copter-44/169

My guess is that the issue is that Lidar provided data that made the object avoidance feature (perhaps BendyRuler was enabled) unable to reach the target waypoint.  Perhaps the WPNAV_RADIUS is set very small?

In any case, this is really a support call so I think I will close this until it's confirmed to be a bug.  Sorry to be strict but if we're not then our to-do list turns into a support forum.

closed 2023-08-25 by rmackay9

Prasanna-icefire, 2023-08-25:
> ## Bug report
> Auto Mission fail with RP Lidar A2 integrated
>
> **Issue details** With the obstacle avoidance parameters enabled and the RPLidar A2 integrated, the drone fails to execute the auto [...]

The drone completes the mission perfectly in auto mode with the avoidance disabled. However on enabling it, it goes round where there are no obstacles. Yes, the bendy ruler and dijktras are enabled simultaneously. There are no obstacles to detect when the drone is in the sky at 10m height. I'll send up the log file later. But thanks for the quick response!

rmackay9, 2023-08-25:
@Prasanna-icefire,

Great.  By the way, the RPLidarA2 performs very badly outdoors because it is very sensitive to sunlight which leads to false positives.

Prasanna-icefire, 2023-08-25:
ohh, makes sense about the false readings. I am not sure if imaginary objects are probably the cause though. It worked perfectly in Loiter mode Imaginary objects would set the drone spinning in loiter as well if it saw the obstacles.

rmackay9, 2023-08-25:
@Prasanna-icefire,

Object avoidance in Loiter is very different from Auto though.  Loiter only uses "simple avoidance" which means it just stops.  Auto uses path planning around obstacles (BendyRuler and Dijkstras).

With false readings, Loiter would tend to stop when it doesn't need to.  It wouldn't "spin" I think.

Prasanna-icefire, 2023-08-25:
The drone in loiter moved away as I approached it. In auto, it revolved in circles without obstacles around.

Loiter behavior:
https://www.instagram.com/p/CwSj5nROXjN/?hl=en

rmackay9, 2023-08-25:
@Prasanna-icefire,

Nice video.

Yes, in Loiter it will backaway because this is part of the Simple avoidance feature.  In Auto it will try to find a path around the obstacle to get to its final target.  Very different behaviour..