ArduPilot/ardupilot #24780: Will not recognize u-Blox GNSS UBX-M9140 SPG 4.04 ROM Based Devices correctly

cturvey, 2023-08-26:
## Bug report

**Issue details**

Code only identifies FLASH based "EXT CORE" SPG 4.xx firmwares, most current step of IC has SPG 4.04 in ROM.
The firmware is near identical save a couple of bytes, and subsequent signing hash, so otherwise commands
and response will work as expected

Discovered during code walk / inspection

Management code for UBX-MON-VER response

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_UBLOX.cpp#L1299

            if (strncmp(_version.hwVersion, "00190000", 8) == 0) {
                ...
                if (strncmp(_version.swVersion, "EXT CORE 4", 10) == 0) {
                    // a M9
                    _hardware_generation = UBLOX_M9;
                }

Needs to also identify "ROM CORE 4"

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**

BEITIAN BK-357 GPS  UBX-M9140 w/SPG 4.04 in ROM, they have several other M9 modules of various sizes also

Startup NMEA sentences, w/u-Center time tags
00:12:46  $GNTXT,01,01,02,u-blox AG - www.u-blox.com*4E
00:12:46  $GNTXT,01,01,02,HW UBX 9 00190000*12
00:12:46  $GNTXT,01,01,02,ROM CORE 4.04 (d964f4)*2A
00:12:46  $GNTXT,01,01,02,FWVER=SPG 4.04*44
00:12:46  $GNTXT,01,01,02,PROTVER=32.01*18
00:12:46  $GNTXT,01,01,02,GPS;GLO;GAL;BDS*77
00:12:46  $GNTXT,01,01,02,SBAS;QZSS*60
00:12:46  $GNTXT,01,01,02,ANTSUPERV=*22
00:12:46  $GNTXT,01,01,02,ANTSTATUS=DONTKNOW*2D
00:12:46  $GNTXT,01,01,02,PF=FFFFB*3A

UBX-MON-VER
0000  B5 62 0A 04 A0 00 52 4F 4D 20 43 4F 52 45 20 34  µb....ROM CORE 4
0010  2E 30 34 20 28 64 39 36 34 66 34 29 00 00 00 00  .04 (d964f4)....
0020  00 00 00 00 30 30 31 39 30 30 30 30 00 00 46 57  ....00190000..FW
0030  56 45 52 3D 53 50 47 20 34 2E 30 34 00 00 00 00  VER=SPG 4.04....
0040  00 00 00 00 00 00 00 00 00 00 00 00 50 52 4F 54  ............PROT
0050  56 45 52 3D 33 32 2E 30 31 00 00 00 00 00 00 00  VER=32.01.......
0060  00 00 00 00 00 00 00 00 00 00 47 50 53 3B 47 4C  ..........GPS;GL
0070  4F 3B 47 41 4C 3B 42 44 53 00 00 00 00 00 00 00  O;GAL;BDS.......
0080  00 00 00 00 00 00 00 00 53 42 41 53 3B 51 5A 53  ........SBAS;QZS
0090  53 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  S...............
00A0  00 00 00 00 00 00 40 41                          ......@A

Thanks, -Clive

shmosta, 2025-02-17:
Hi everyone, this bug seems to affect all Emlid Navio2 hats.  Last version working for my Navio2 was ardupilot 4.4.4.  I'm going to try to hack a work-around for this, and if you have any tips or suggestions on what to change, I would be most appreciative.  Thanks!

cturvey, 2025-02-27:
Code's moved down here now,  https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_UBLOX.cpp#L1450

I'd just compare both patterns, super low risk
```
                if ((strncmp(_version.swVersion, "EXT CORE 4", 10) == 0) ||  // FLASH variant of SPG 4.0x w/ROM 1.0x
                    (strncmp(_version.swVersion, "ROM CORE 4", 10) == 0)) { // ROM variant of SPG 4.04
                    // a M9 w/UBX-M9140 IC
                    _hardware_generation = UBLOX_M9;
                }
```
-Clive  (clive1 at u-Blox forums)