ArduPilot/ardupilot #24798: ESC_TELEM_MAX_ESCS <=3 never sends MSG_ESC_TELEMETRY_1_TO_4

magicrub, 2023-08-28:
ESC_Telemetry will not send MAVLink MSG_ESC_TELEMETRY_1_TO_4 if ESC_TELEM_MAX_ESCS (PWM/SERVO) count is <=3

https://github.com/ArduPilot/ardupilot/blob/dc8455411adb90308eff08b47d3db7c4faf7be1e/libraries/AP_ESC_Telem/AP_ESC_Telem.cpp#L312C50-L312C50

```
const uint8_t num_idx = ESC_TELEM_MAX_ESCS/4;
    for (uint8_t idx = 0; idx < num_idx; idx++) {
        const uint8_t i = (next_idx + idx) % num_idx;
        ...
```

There's actually two problems caused by `num_idx == (ESC_TELEM_MAX_ESCS/4) == 0`:
1. the for loop will loop 0 times (will never send anything)
2. `foo % num_idx` throws a compiler warning/error about preforming mod 0 which is a divide-by-zero (but unreachable).

I made PR https://github.com/ArduPilot/ardupilot/pull/24771 to fix the divide-by-zero but @andyp1per noticed the will-never-send bug so I closed it in favor of a more complete solution.

leviosacz, 2023-08-29:
Hello!

I'm interested in addressing this issue. Currently, the code processes ESCs in groups of 4. However, there's an observation that the last group, with fewer than 4 ESCs, isn't processed. This becomes apparent when considering cases like having 7 ESCs, where the last group containing only 3 ESCs is left unprocessed.

I'm seeking clarity on the intended behavior for these cases. Is there an intention to address the last group issue, treating it as if it had 3 ESCs, or should special handling be implemented specifically when `ESC_TELEM_MAX_ESCS` (PWM/SERVO count) is equal to or less than 3?

adityaomar3, 2023-09-07:
Update logical error in grouping of 4 escs in AP_ESC_Telem.cpp #24912
gh pr checkout 24912
I have tried to solve the issue by changing the logic a bit for grouping of 4 escs.
The logical error was arising due to grouping of 4 by division it may lose some values if total escs are not the multiple of 4 so I have updated it accordingly.
Please review it @magicrub

peterbarker, 2024-07-30:
This was fixed by @adityaomar3 - thanks!

closed 2024-07-30 by peterbarker