ArduPilot/ardupilot #24809: cmake build failed

keyss-abhay, 2023-08-29:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

keyss-abhay, 2023-08-29:
Hello everyone
I am running the build of Ardupilot 
[image]

I am trying to resolve the issues of make

khancyr, 2023-08-29:
Hello,

We don't support Cmake. 
Please follow instruction from https://ardupilot.org/dev/docs/building-the-code.html to build a firmware.
Thanks

closed 2023-08-29 by khancyr