ArduPilot/ardupilot #24836: Plane: TECS update and usage race condition resulting in NaN

IamPete1, 2023-08-31:
It is possible to try and run TECS having never told it the airspeed. In the worst case this can result in it trying to use 0 and getting a constraining NaN internal error (I have a log of this).

The a 50Hz loop checks if it should run TECS here:
https://github.com/ArduPilot/ardupilot/blob/ad19321d8974d2906607fed2142cd96a5368c6a5/ArduPlane/ArduPlane.cpp#L207

That if that returns false so TECS is not run.

 Now we change modes and check again here in a different loop run at 10Hz rate:
https://github.com/ArduPilot/ardupilot/blob/ad19321d8974d2906607fed2142cd96a5368c6a5/ArduPlane/ArduPlane.cpp#L554

Now because we changed modes in between we have not run `TECS_controller.update_50hz()` recently (or at all) so when we call `TECS_controller.update_pitch_throttle` it has a out of date speed.

The way this causes the internal error is if the user has set `INITIAL_MODE` to a mode that does not run TECS so its never run, and the airspeed is never set to a value `update_50hz()` constraint is to be at least 3m/s. But is TECS has never run its still 0. That 0 propagates from `_TAS_state` to the `gainInv` here:
https://github.com/ArduPilot/ardupilot/blob/ad19321d8974d2906607fed2142cd96a5368c6a5/libraries/AP_TECS/AP_TECS.cpp#L984

Which is then used here resulting in a NaN:
https://github.com/ArduPilot/ardupilot/blob/ad19321d8974d2906607fed2142cd96a5368c6a5/libraries/AP_TECS/AP_TECS.cpp#L1027

That is then constrained here:
https://github.com/ArduPilot/ardupilot/blob/ad19321d8974d2906607fed2142cd96a5368c6a5/libraries/AP_TECS/AP_TECS.cpp#L1035

And we get the internal error. The `TECS` log shows the 0 value of `_TAS_state` in the `sp` field:

[image]

The next time we come through that has been constrained up to that minimum 3 m/s value. 

It would be quite easy to paper over the internal error with a extra constrain, but that does not fix the fundamental issue of using stale data. The only way to fix that is to fix the race condition between the running and updating of TECS. It looks like the only reason were not running the `TECS_controller.update_50hz()` on every loop is to provoke a reset, the fix would be to run it all the time and add new hooks to trigger the reset.

IamPete1, 2023-09-10:
Closed by https://github.com/ArduPilot/ardupilot/pull/24916

magicrub, 2023-12-12:
I still hit this in Plane 4.4.4beta1 on line:
 `_pitch_dem = constrain_float(_pitch_dem_unc, _PITCHminf, _PITCHmaxf);`

magicrub, 2023-12-12:
bench testing logs: 
[constrain_NaN.zip](https://github.com/ArduPilot/ardupilot/files/13653586/constrain_NaN.zip)