ArduPilot/ardupilot #24855: Change the way horizon line is drawned on OSD when inverted to conform with aviation standards

dxs94, 2023-09-02:
## Feature request

**Is your feature request related to a problem? Please describe.**
When flying inverted (ie after a 180° roll), the artificial horizon pitch indication on the osd is reversed compared to any full size aviation ADI.
To reproduce: fly inverted (half roll). Push on the elevator stick so that airplane is now in a climb with the nose above horizon. Observe that the osd artificial horizon bar moves DOWN in your goggles whereas it should move UP.
Below: video of arduplane behaviour, and pictures of what it should look like.

https://www.youtube.com/watch?v=Jwwi2szEIdQ

[image]
[image]

**Describe the solution you'd like**
Taking into accound b/w character based OSD limitations, something like this would be appropriate:
[image]
while always keeping the horizon line in view even at extreme pitch angles so that the pilot always knows it's attitude.

**Describe alternatives you've considered**
With current OSD limitations none that I can think of.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
An artificial horizon has to present the pilot a virtual horizon line from the pilot’s perspective. That and a sky/ground reference to show which side of the line is earth and which is sky. The whole point is that to level off the aircraft the pilot flies to the ADI horizon line no matter if the plane is upright or inverted. Which is impossible if the ADI is just showing « is the aircraft nose above horizon or not ? » as Ardupilot depicts (on top of not having a sky/ground reference).

IamPete1, 2023-09-03:
This is the code in that would have to be fixed:

https://github.com/ArduPilot/ardupilot/blob/9d98244730cdadf5b1eb8401efd5e641e4099b9b/libraries/AP_OSD/AP_OSD_Screen.cpp#L1570

adityaomar3, 2023-09-07:
understood the issue, I am interested in solving the issue please help me with some jargons like ocd and adi

adityaomar3, 2023-09-07:
hey @IamPete1 I understood the issue and will try to solve it. Can you please help me with the fn 'SYMBOL' and https://github.com/ArduPilot/ardupilot/blob/9d98244730cdadf5b1eb8401efd5e641e4099b9b/libraries/AP_OSD/AP_OSD_Screen.cpp#L1577C7-L1577C7 in this line why pitch = -pitch is done?

adityaomar3, 2023-09-09:
@dxs94 can you please help to understand the issue better so that I can try on resolving it. Correct me if I am wrong the behaviour of osd is opposite on upright and inverted, it should be reversed?

Venkatesan-M, 2023-09-09:
@dxs94 Correct me if I am wrong you want the horizon also to be inverted when the craft is inverted. @IamPete1 I Think there should be a small correction in if (check_option(AP_OSD::OPTION_INVERTED_AH_ROLL)) part in the above code you have mentioned. is the issue still open, Can I go ahead and make a PR?

dxs94, 2023-09-09:
Thanks for dev-capable people to look into this. I tried to make the post as clear as possible, please carefully check the YT video (showing incorrect behavior), and the correct behavior on the TX16S script. Below that is my proposal with some « V » to differentiate sky side from ground side of the horizon line, maybe there is a better way depending on available processing power / osd bandwidth. This shows what the OSD would depict if code was correct when aircraft is rolling upside - down (ie. making a half roll).

Fundamentally the problem lies in the fact that Ardupilot code has a pitch value of say +20° when aircraft nose is above horizon. This doesn’t take roll angle into account. And the OSD drawing routine always puts the horizon line below aircraft reference symbol in this case because pitch value hasn’t changed and is positive, even when aircraft is now flying inverted.
To correct this you have to make cases for when roll angle is >90° or <-90°.

Hwurzburg, 2023-09-09:
It depends on how you want to represent the display....as it is now the target mark represents the nose of the aircraft (always in the center of the OSD) and the horizon moves appropriately (and correctly even when inverted) ...AND matches the horizon you see when flying FPV (very important)...I fly inverted often (and am a full size pilot) and prefer knowing where my nose is with respect to the horizon...IF we had a color display such that earth vs sky could be depicted then a picture like the horizon in a real plane could be viable....flying FPV having the horizon move away from what is in the video view would be very distracting....since our pitch information (-90 to +90)cannot be used to determine inversion (must be calculated versus current roll angle which is -180 to 180) the above PR does not work and is wrong when flying upright also...I am not opposed to coding a display as requested as an option (like the inverted_roll option, which tried to come closer to the Russion HUD roll presentation)

adityaomar3, 2023-09-09:
@dxs94 sorry for not asking the right question, I mean to ask that if plane is upright then its nose should be above horizon and vice versa ? Whats wrong with that?
Sorry if you find my question unworthy, I am newbie here 😅

Hwurzburg, 2023-09-09:
BTW...the screen shots of the Yaapu OpenTX display you show have nothing to do with ArduPilot's OSD HUD display...they are generated locally from attitude information sent over telemetry....changing the AP OSD would have absolutely no impact on the TX display you show....for that, you would need to have the Yaapu script changed....and for that option, since the Horus class TXs DO have color display, it might be a worthwhile change...I suggest you open a feature request in Yaapu's git repo

dxs94, 2023-09-10:
@Hwurzburg @adityaomar3 may I kindly ask you re-check the first feature post? I updated it with better wording, the Yaapu screenshots ARE CORRECT and the Yaapu script certainly does NOT need to be changed. And yes I get it only displays telemetry data.

Here is a picture of what the OSD horizon should look like (and would like in a full size aircraft), based on the still pictures you posted just above. Because the whole point of the horizon line is to represent the horizon when you can't see it (ie. in clouds).
[image]

To make things clear please review this: The same picture, but you just entered a cloud. How do you know where earth is? How do you know if you're inverted or not? How do you know if you have to pull or push on the elevator to level the aircraft? You can't.
[image]

Now check this one. Here I can tell I'm upside down (thanks to the chevrons on horizon line), AND I can level the aircraft **by flying towards the horizon line**. The fact I'm upside down doesn't matter because the horizon line on the OSD is **at the same position as the REAL horizon line**.
[image]

Hwurzburg, 2023-09-10:
okay I have coded the mode option and verified in SITL....will post a video later today here....and I added an indication to tell when inverted....flying it under the hood without FPV I can see the advantage, although when overlaid with video it would take a little getting used to....I will have to fly it some to decide if this option should be defaulted on or off...

Hwurzburg, 2023-09-10:
https://youtu.be/RZamcZy_5Zs

dxs94, 2023-09-10:
Nice work that was fast! Spot on.
Have you tried instead of the blinking, using characters like this to depict the horizon line: « V - V - - - - V - V »? This would be with the sky at the lower tip of the « V » and the ground on the side the « V » has 2 branches. After some flights I would be more intuitive than blinking (in my opinion).
Anyway thank you for your work.

Hwurzburg, 2023-09-10:
> Nice work that was fast! Spot on. Have you tried instead of the blinking, using characters like this to depict the horizon line: « V - V - - - - V - V »? This would be with the sky at the lower tip [...]

changing characters could have bad impacts on all the different MSP displays and fonts sets being used....this panel is drawn in all  DisplayPort variations...the letter "v" is implemented as italics in some sets, with cursive in others, shadowed in others, customized in others......better stick with the icon number everyone uses for the horizon line rather than change it

dxs94, 2023-09-10:
Ok I have no knowledge about those constraints, the blinking does the job just fine.
Looking forward to flying with this new option! Thanks!

Hwurzburg, 2023-09-12:
option merged into master...also added blinking horizon for FPV style (default)

closed 2023-09-12 by Hwurzburg