ArduPilot/ardupilot #24862: ap_dds_tests: Deprecate calling setup.py directly

Ryanf55, 2023-09-03:
## Bug report

**Issue details**

Developer machine 

**Version**
master 9d98244730cdadf5b1eb8401efd5e641e4099b9b

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**

```
colcon build --packages-up-to ardupilot_dds_tests
...
Starting >>> ardupilot_dds_tests
--- stderr: ardupilot_dds_tests                                                                        
/home/ryan/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< ardupilot_dds_tests [0.54s]

```

Tagging @srmainwaring  for visibility

srmainwaring, 2023-09-03:
This requires an upstream fix to colcon. You see this on all the ROS 2 projects built as ament_python packages in humble built with Python versions >= 3.10.

There is discussion here:
- https://github.com/colcon/colcon-core/issues/454#issuecomment-1262592774
- https://discourse.ros.org/t/infrastructure-community-meeting-2023-05-17/31268

Call for testing the new approach here:
-https://discourse.ros.org/t/call-for-testing-standards-based-python-packaging-with-colcon/32008

vibgyor-s, 2023-09-11:
Reporting that after the setuptools update to version 68.2.0 [changelogs](https://setuptools.pypa.io/en/latest/history.html) the package straightaway fails to build with error **Invalid Syntax**

> Starting >>> ardupilot_dds_tests
--- stderr: ardupilot_dds_tests                   
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/usr/lib/python3/dist-packages/colcon_ros/package_augmentation/ros_ament_python.py", line 57, in getter
    return get_setup_information(
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 293, in _get_setup_information
    return ast.literal_eval(output)
  File "/usr/lib/python3.10/ast.py", line 64, in literal_eval
    node_or_string = parse(node_or_string.lstrip(" \t"), mode='eval')
  File "/usr/lib/python3.10/ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 1
    {'package_data': {}, 'dist_files': [], 'src_root': None, 'dependency_links': [], 'setup_requires': [], 'eager_resources': None, 'entry_points': {'console_scripts': ['time_listener = ardupilot_dds_tests.time_listener:main']}, 'exclude_package_data': None, 'extras_require': {}, 'include_package_data': None, 'install_requires': ['setuptools'], 'namespace_packages': None, 'packages': ['ardupilot_dds_tests'], 'python_requires': None, 'test_loader': None, 'test_runner': None, 'test_suite': None, 'tests_require': ['pytest'], 'use_2to3': None, 'zip_safe': True, 'message_extractors': None, 'verbose': 1, 'dry_run': 0, 'help': 0, 'command_packages': None, 'script_name': 'setup.py', 'script_args': ['--dry-run'], 'command_options': {'develop': {'script_dir': ('setup.cfg', '$base/lib/ardupilot_dds_tests')}, 'install': {'install_scripts': ('setup.cfg', '$base/lib/ardupilot_dds_tests')}}, 'package_dir': {}, 'py_modules': None, 'libraries': None, 'headers': None, 'ext_package': None, 'include_dirs': None, 'extra_path': None, 'scripts': None, 'data_files': [('share/ament_index/resource_index/packages', ['resource/ardupilot_dds_tests']), ('share/ardupilot_dds_tests', ['package.xml']), ('share/ardupilot_dds_tests/launch', []), ('share/ardupilot_dds_tests/config', []), ('share/ardupilot_dds_tests/config', ['config/ardupilot_dds.yaml'])], 'password': '', 'command_obj': {}, 'have_run': {}, 'want_user_cfg': True, 'metadata': {'name': 'ardupilot_dds_tests', 'version': '0.0.0', 'author': None, 'author_email': None, 'maintainer': 'maintainer', 'maintainer_email': 'maintainer@ardupilot.org', 'url': None, 'license': 'GPL-3.0', 'description': 'Tests for the ArduPilot AP_DDS library', 'long_description': None, 'keywords': None, 'platforms': None, 'classifiers': None, 'download_url': None, 'provides': None, 'requires': None, 'obsoletes': None, 'long_description_content_type': None, 'project_urls': {}, 'license_file': None, '_normalized_install_requires': {'setuptools': <Requirement('setuptools')>}, '_normalized_extras_require': {}}}
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            [... cut ...]                                                                                                                                           ^
SyntaxError: invalid syntax
Failed   <<< ardupilot_dds_tests [0.21s, exited with code 1]
Summary: 3 packages finished [7.60s]
  1 package failed: ardupilot_dds_tests
  1 package had stderr output: ardupilot_dds_tests
invalid syntax (<unknown>, line 1)

---

**This is verified after completing the build by reverting to setuptools==68.1.2 as a temporary fix.**