ArduPilot/ardupilot #24865: narrowing conversion of 'fmax

lixiaowei123, 2023-09-04:
[image]

/../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp: In member function 'void NavEKF3_core::send_status_report(GCS_MAVLINK&) const':
../../libraries/AP_Math/ftype.h:24:24: error: narrowing conversion of 'fmax(fmax(((double)magVar.Vector3<float>::x), ((double)magVar.Vector3<float>::y)), ((double)magVar.Vector3<float>::z))' from 'double' to 'float' inside { } [-Werror=narrowing]
 #define fmaxF(x,y) fmax(x,y)
                        ^
../../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp:620:9: note: in expansion of macro 'fmaxF'
         fmaxF(fmaxF(magVar.x,magVar.y),magVar.z),

IamPete1, 2023-09-05:
What is the output of your configure step?

peterbarker, 2023-09-08:
I think this looks like a valid warning; the `mavlink_ekf_status_report_t` we are assembling really only has room for a float but we're trying to shove a double in there (`fmaxF` sometimes doing that....).

We should not be using `fmaxF` there - the type is clearly float IMO.  https://github.com/ArduPilot/ardupilot/pull/24918

I am *really* interested in the output of that configure step which @IamPete1 requested!

lixiaowei123, 2023-09-14:
thank you,Gcc version issue

IamPete1, 2023-09-27:
Fixed by https://github.com/ArduPilot/ardupilot/pull/24918

closed 2023-09-27 by IamPete1