ArduPilot/ardupilot #24874: this is my issue Got COMMAND_ACK: VIDEO_START_CAPTURE: FAILED and Got COMMAND_ACK: VIDEO_STOP_CAPTURE: FAILED and i am using arducopter v4.5.0 version could you please tell me solution

MalleswariReddy1593, 2023-09-05:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

closed 2023-09-05 by amilcarlucas

rmackay9, 2023-09-05:
I agree that this is best raised in the forums (e.g. Copter-4.4 is a good choice even though you're on 4.5.0, https://discuss.ardupilot.org/c/arducopter/copter-44/169).  Please include a log of course.

My guess is that the CAM1_TYPE parameter has not been set to "4" (Mount) but this really is just a guess.

MalleswariReddy1593, 2023-09-05:
still facing same issue this is my issue Got COMMAND_ACK: VIDEO_START_CAPTURE: FAILED and Got COMMAND_ACK: VIDEO_STOP_CAPTURE: FAILED please help me

rmackay9, 2023-09-05:
@MalleswariReddy1593,

As mentioned above, let's keep the discussion in the forum and please include a log.  If there's no log, I just don't have the information I need to help you.