ArduPilot/ardupilot #24879: Cannot locate rosdep definition for [ardupilot_sitl]

SantamRC, 2023-09-05:
## Bug report

When executing `rosdep install --rosdistro ${ROS_DISTRO} --from-paths src` from [this](https://ardupilot.org/dev/docs/ros2.html) documentation, the following error was encountered,

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ardupilot_dds_tests: Cannot locate rosdep definition for [ardupilot_sitl]
```

**Version**

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

@Ryanf55 please have a look at it.

Ryanf55, 2023-09-05:
Confirmed in a ROS 2 humble docker container on `master` hash 26b7092c1ac6acc55d2d455d36d3f60dae38cea4

ojas2412, 2024-03-03:
Hello! Is anyone working on this currently?

Ryanf55, 2024-03-03:
Not now. I've assigned it to you.

ojas2412, 2024-03-07:
Hey @Ryanf55. Tried to replicate this on an Ubuntu 20.04 system I've got running, a few impressions:
1. The documentation for ROS2 installation from source given above no longer instructs running `rosdep install --rosdistro ${ROS_DISTRO} --from-paths src`. I understand this issue is a bit dated, have changes been made to the original wiki since the creation of this issue?

2. Tried to replicate a on a fresh (no prior packages installed apart from the ones listed on the ardupilot setting-up wiki) Ubuntu Focal machine. However, I didn't encounter this error, even with `master` hash `26b7092`. Could you help me out with this?

ojas2412, 2024-03-07:
I did, however, notice the following issues in the overall process of creating a ROS2 workspace, building ArduPilot from scratch, and installing ROS2 as required for DDS dependencies:

1. In the `rosdep install --from-paths src --ignore-src` section in [this](https://ardupilot.org/dev/docs/ros2.html) tutorial, I received the following errors:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
```
micro_ros_agent: No definition of [micro_ros_msgs] for OS version [focal]
ardupilot_dds_tests: No definition of [micro_ros_msgs] for OS version [focal]
ardupilot_sitl: No definition of [ament_cmake_black] for OS version [focal]
```
Digging a bit further, I read that rosdep keys for these packages aren't natively available for Ubuntu 20.04, since they recently moved these keys to Ubuntu 22.04 as the preferred support machine for ROS Humble, Iron etc.

Solved this by directly cloning and building these modules into the workspace's `src`. From what I tried, the [build dependencies for DDS](https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_DDS#installing-build-dependencies) couldn't solve this either. 

Could you verify if this is an issue you would like fixed? If so, I can try a PR where I've thought of adding these 3 modules to `ardupilot/master/Tools/ros2/ros2.repos` itself. `wget` can directly eliminate this hassle then.

2. `colcon build` finished, but with a `stderr` output for `ardupilot_dds_tests`:

```
--- stderr: ardupilot_dds_tests                     

=============================== warnings summary ===============================
../../../../../../../../usr/lib/python3/dist-packages/pydocstyle/config.py:6
  Warning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.10 it will stop working

```
Would you like to have this collections method updated? Please do let me know.

Ryanf55, 2024-03-08:
> I did, however, notice the following issues in the overall process of creating a ROS2 workspace, building ArduPilot from scratch, and installing ROS2 as required for DDS dependencies:
>
> 1. In the [...]

Hello.

The main problem is you are using the wrong ubuntu version. 
https://docs.ros.org/en/humble/Installation.html

If you are able to repeat the tests on the correct OS, I would be interested to know if this is still a problem.

Ryanf55, 2024-03-11:
I've tried in a fresh workspace, this is no longer an issue if you follow the wiki instructions:
https://ardupilot.org/dev/docs/ros2-sitl.html

usmanNoor5, 2024-09-29:
i am new to it having some issues
rosdep install --rosdistro humble --from-paths src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ardupilot_gz_gazebo: Cannot locate rosdep definition for [gz-sim8]
ardupilot_sitl_models: Cannot locate rosdep definition for [ardupilot_gazebo]
ardupilot_dds_tests: Cannot locate rosdep definition for [ardupilot_sitl]
ardupilot_sitl: Cannot locate rosdep definition for [micro_ros_agent]
ardupilot_gz_bringup: Cannot locate rosdep definition for [ardupilot_sitl_models]
ardupilot_ros: Cannot locate rosdep definition for [ardupilot_sitl]
ardupilot_gazebo: Cannot locate rosdep definition for [gz-sim8]