ArduPilot/ardupilot #24894: RSSI reading from CRSF receiver protocol incorrect

RBirdie001, 2023-09-06:
## Bug report

**Issue details**

When using Crossfire or ELRS receiver connected via CRSF serial protocol, RSSI reading to both OSD and SD card log is incorrect. RSSI values looks too much similar to LQ readout, just in different scale. Compared with Edge TX log showing correct curves. All logs attached under the link below. RSSI graph from Edge is logaritmic, but its basically different shape is obvious.
Due to different scale, I had to overlay RSSI and LQ outputs from Mission Planner log viewer as images.

**Version**
Arduplane 4.4

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Airplane V tail

**Hardware type**
Omnibus F4 V3

**Logs**

[image]
[image]
(https://drive.google.com/drive/folders/1sDo8mVCQYkAtxfW-iOQ-pwkiOvT4UvvB?usp=sharing)

andyp1per, 2023-09-07:
There is no true RSSI value on these platforms it is just an approximation based on TBS recommendation

kolins-cz, 2023-09-07:
> There is no true RSSI value on these platforms it is just an approximation based on TBS recommendation

What are you saying? CRSF protocol returns true RSSI in dBm and Link Quality in percent. It is clearly stated in **CRSF Rev07.pdf**

Or by these platforms you mean Arducopter/Arduplane does not support true RSSI?

 
**Snippet from CRSF protocol documentation:**
0x14 Link statistics
Payload:
● uint8_t Uplink RSSI Ant. 1 ( dBm * -1 )
● uint8_t Uplink RSSI Ant. 2 ( dBm * -1 )
● uint8_t Uplink Package success rate / Link quality ( % )
● int8_t Uplink SNR ( db )
● uint8_t Diversity active antenna ( enum ant. 1 = 0, ant. 2 )
● uint8_t RF Mode ( enum 4fps = 0 , 50fps, 150hz)
● uint8_t Uplink TX Power ( enum 0mW = 0, 10mW, 25 mW, 100 mW, 500
mW, 1000 mW, 2000mW )
● uint8_t Downlink RSSI ( dBm * -1 )
● uint8_t Downlink package success rate / Link quality ( % )
● int8_t Downlink SNR ( db )
Uplink is the connection from the ground to the UAV and downlink the opposite direction.
[CRSF Rev07.pdf](https://github.com/ArduPilot/ardupilot/files/12546402/CRSF.Rev07.pdf)

jiri-jirus, 2023-09-07:
nevertheless LQ in Arducopter should take its data from CRSF's RQLy rather than calculating its RSSI.

RBirdie001, 2023-09-07:
Yes, it's exactly what I was looking for! Now I can see it even in "all parameters" list but I'd bet it wasn't there before... Was it updated recently?
Of course, possibility to show real dBm value would be even better, but this will help me.

closed 2023-09-07 by RBirdie001