ArduPilot/ardupilot #24909: check if the protocol works

PR-JHL, 2023-09-07:
https://github.com/PR-JHL/ardupilot_202308/blob/3f2ef4df7d978d6efe7e6f3809ca909ec0180a12/libraries/AP_SerialManager/AP_SerialManager.cpp#L538

amilcarlucas, 2023-09-07:
You need to explain what the problem is. We can not guess it.

peterbarker, 2023-09-08:
As luck would have it, we recently merged a LUA script to run DLA ECUs: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/drivers/EFI_DLA.lua

If you do want to run a .cpp driver for this ECU then you will want to put it in with the existing `AP_EFI` library rather than as a separate serial protocol.

PR-JHL, 2023-09-08:
Oh thanks! I would try that. By the way, I'm not using a recommended DLA232. The current ECU uses a ‘exclusive' (as the customer service told me) protocol with CRC. So I guess I need to do some modification?

peterbarker, 2024-07-31:
Closing this issue - yes, if you want to support a new protocol, modifications will be needed :-)

closed 2024-07-31 by peterbarker