ArduPilot/ardupilot #24931: Copter firmware MatekF405-TE-bdshot on FC "Matek F405-miniTE" does not initialize Dshot on Servo3

dejero13, 2023-09-09:
## Bug report

**Issue details**
I wanted to use four motors on the copter with BDshot, but the firmware MatekF405-TE-bdshot on FC "Matek F405-miniTE" does not initialize Dshot on Servo3 (regardless of servo_blh _bdmask), there are no problems with Servo1,2,4. With firmware MatekF405-TE all Servo1,2,3,4 work, but there is no BDshot. I also checked the FC on the Betaflight firmware, where all 4 motors work with BDshot.
I attach a screenshot of the MissionPlanner messages and a param file.

**Version**
MatekF405-TE-bdshot 4.4.0

**Platform**
[  ] All
[  ] AntennaTracker
[+] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Matek F405-miniTE

**Attach**
[image]

[MatekF405-TE-bdshot_4.4.0.param.zip](https://github.com/ArduPilot/ardupilot/files/12566460/MatekF405-TE-bdshot_4.4.0.param.zip)

dejero13, 2023-09-09:
On the Mateksys website there is a note to this FC with a similar solved problem for Inav:
"INAV firmware, S3, S5,S7 DSHOT not working because of DMA clash,  pls use ONESHOT or MULTISHOT and calibrate ESC PWM range.
This DMA clash will be fixed from INAV 7.x.x."

[FLIGHT CONTROLLER F405-MINITE](http://www.mateksys.com/?portfolio=f405-minite)

dejero13, 2023-09-09:
Perhaps this Inav topic has just solved the current problem.
[Timer N channel fix](https://github.com/iNavFlight/inav/pull/9276)

andyp1per, 2023-09-09:
This is by design - PWM3 uses a timer that makes bdshot not possible - you should use another channel for a motor

closed 2023-09-09 by andyp1per