ArduPilot/ardupilot #24941: ArduCopter: Terrain Check request leads to reappearance of deleted FATFS contents

sahilsihag, 2023-09-11:
## Bug report
**Issue summary**  
A TERRAIN_CHECK (135) request for a random latitude, longitude leads to appearance of a file. This file contains data that has been earlier deleted but not overwritten from the microSD card (FATFS). The size of file can be influenced by changing values of latitude, longitude.

**Issue details**  
```
MAVLink packet: fd030000a7ffe6870000244aca85cb

Information:
Payload: TERRAIN_CHECK (135)
    lat (int32_t): 13257252
    lon (int32_t): 0
```
leads to `case DiskIoWaitRead` ([libraries/AP_Terrain/TerrainIO.cpp](https://github.com/ArduPilot/ardupilot/blob/502702df62572519b56971fe9fed5b2883640879/libraries/AP_Terrain/TerrainIO.cpp#L370)). 

```cpp
    case DiskIoWaitRead:
        // need to read in the block
        open_file();
                                         // inside open_file()
                                         // AP::FS().open(file_path, O_RDWR|O_CREAT); <-- tries opening "/APM/TERRAIN/N01E000.DAT"
                                         // Open is successful
        if (fd == -1) {
            return;
        }
        read_block();
                                         // inside read_block()
                                         // AP::FS().lseek(fd, file_offset, SEEK_SET) <-- tries to lseek to file_offset (calculated from latitude, longitude)
                                         // lseek is successful

                                         // AP::FS().read(fd, &disk_block, sizeof(disk_block)) <-- tries to read after reaching offset
                                         // but now `ret != sizeof(disk_block)` is True 
                                         // read is unsuccessful. Enabling `TERRAIN_DEBUG` leads to "read empty block ..."
        break;
```
If we now query filesystem via mavproxy, we see:
```
ftp list /APM/TERRAIN                        
STABILIZE> Listing /APM/TERRAIN                
   N01E000.DAT  0                            
Total size 0.00 kByte  
```
`fd` is not closed yet in the expectation of being reused. So, we send a different packet:
```
MAVLink packet: fd0400000678e687000084ada60f7e9c

Information:
Payload: TERRAIN_CHECK (135)
    lat (int32_t): 262581636
    lon (int32_t): 0
```
this again leads to `case DiskIoWaitRead` ([libraries/AP_Terrain/TerrainIO.cpp](https://github.com/ArduPilot/ardupilot/blob/502702df62572519b56971fe9fed5b2883640879/libraries/AP_Terrain/TerrainIO.cpp#L370)). 

```cpp
    case DiskIoWaitRead:
        // need to read in the block
        open_file();
                                         // But now there is different story
                                         // inside open_file()
                                         // AP::FS().close(fd); <-- closes old fd
                                         // rest does not matter for this bug
```
If we now query file system again via mavproxy, we see:
```
ftp list /APM/TERRAIN    
STABILIZE> Listing /APM/TERRAIN
   N01E000.DAT  1290240  
   N26E000.DAT  0  
Total size 1260.00 kByte  
```
The size of `N01E000.DAT` is same as `file_offset` used for `AP::FS().lseek`. The content appearing inside this file match with dump of microSD card taken before doing this experiment (`dd if=/dev/mmcblk0p1 of=dump_file status=progress`) .

Later we also see:
```
ftp list /APM/TERRAIN    
STABILIZE> Listing /APM/TERRAIN     
   N01E000.DAT  1292288    
   N26E000.DAT  856064      
Total size 2098.00 kByte 
```
This is because mavproxy sends TERRAIN_DATA (134). This data gets written to the end of file as expected.

**Additional Comments**  
When the above operation tries to `lseek` it calls `f_lseek` ([ardupilot/modules/ChibiOS/ext/fatfs/source/ff.c](https://github.com/ArduPilot/ChibiOS/blob/1ec9f168b2e64a043470a87225b405e79d187354/ext/fatfs/source/ff.c#L4423)). There are hints of file size being zero (`fp->obj.objsize` is 0) and [if condition](https://github.com/ArduPilot/ChibiOS/blob/1ec9f168b2e64a043470a87225b405e79d187354/ext/fatfs/source/ff.c#L4504)
```cpp
		if (ofs > fp->obj.objsize && (FF_FS_READONLY || !(fp->flag & FA_WRITE))) {	/* In read-only mode, clip offset with the file size */
			ofs = fp->obj.objsize;
		}
```
to restrict offset (`ofs`) to `fp->obj.objsize`. But this never happens in above situation, possibly due to `FF_FS_READONLY` not being True.

**Version**  
ArduCopter V4.4.0 (502702df)

**Platform**  
[ ] All
[ ] AntennaTracker
[x] Copter -> **with microSD card (FATFS)**
[ ] Plane
[ ] Rover
[ ] Submarine

**Airframe type**  
Not applicable.

**Hardware type**  
Board: omnibusF4
With microSD card (FATFS)

**Logs**  
Already inline above.

tridge, 2023-12-11:
yes, we use sparse files deliberately in AP_Terrain, and we use a header in the file that indicates which blocks are valid to use. We deliberately don't fill in the unused blocks

sahilsihag, 2023-12-11:
I might be mistaken, but I am referring to a different situation altogether. What I would like to convey:
- With a `TERRAIN_CHECK` message, IO thread can be forced to read a file from disk which does not even exist (case [`DiskIoWaitRead`](https://github.com/ArduPilot/ardupilot/blob/502702df62572519b56971fe9fed5b2883640879/libraries/AP_Terrain/TerrainIO.cpp#L370))
- Even then, the firmware tries to lseek file descriptor (fd) to offset specified in TERRAIN_CHECK message.
- When this offsetted fd is closed, a file appears in file system (here: `N01E000.DAT`) which exposes bytes directly from the device hardware as a file system entry.

ChibiOS allows restricting file's lseek offset to its size [here](https://github.com/ArduPilot/ChibiOS/blob/1ec9f168b2e64a043470a87225b405e79d187354/ext/fatfs/source/ff.c#L4504) but ArduPilot does not make use of this condition.

As a result: microSD card can be physically filled with only a few malicious/wrong TERRAIN_CHECKs and expose files which were deleted but not wiped at device block level.

tridge, 2023-12-15:
> As a result: microSD card can be physically filled with only a few malicious/wrong TERRAIN_CHECKs and expose files which were deleted but not wiped at device block level.

If someone malicious controls the contents of the sdcard then the vehicle is completely compromised. That is not an interesting case to consider.
For the non-malicious case the blocks would have to have both a correct CRC, have correct lat/lon, correct spacing, correct version and have the right bitmap bits filled in. So essentially it would need to be valid terrain data.
I'm struggling to see what the real risk is here. The reason we quite deliberately use sparse files like this is the IO cost of pre-filling these files would be huge. The filesystem is slow, so we rely on the signatures in the read blocks to ensure we are getting valid data.
It total we check 14 bytes of data on a block before using it. The chances those 14 bytes are randomly correct is vanishingly small.

rmackay9, 2023-12-15:
I think I will go ahead and close this one as well.  We can re-open of course if this is a mistake but I suspect these issues are really questions and not issues we've seen in practice.

closed 2023-12-15 by rmackay9