ArduPilot/ardupilot #24965: The LO Race install command via Smart Audio does not work on the Rush Max Solo

daniel-yavorovich, 2023-09-12:
## Bug report

The LO Race install command via Smart Audio does not work on the Rush Max Solo.

All other frequencies are set correctly.

```
VTX_BAND=5
VTX_CHANNEL=0
VTX_OPTIONS=8 (Unlocked)
```

With such a configuration, Max Solo stops responding to SmartAudio commands at all (it does not change power, for example).
It is worth replacing VTX_BAND with A Band - how everything starts to work.

**Version**
Stable

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2023-09-12:
Can you provide more details - I'm really not sure what you are asking for here

daniel-yavorovich, 2023-09-12:
@andyp1per sorry for confuse.

I mean I have problem using `VTX_BAND` and `VTX_CHANNEL` parameters when it comes to `LO Race` Band.

At the code level it is described here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_VideoTX.cpp#L99

I have **Rush Max Solo 2.5Wt** which supports Smart Audio protocol. I have the ability to set VTX Power and set `VTX_BAND` to anything but 5 (`LO Race`) working. It looks like some problems at the implementation level of SmartAudio Serial Protocol (because in Betaflight the same functionality works fine).

I hope I managed to describe the problem in more detail, but I will be happy to answer clarifying questions.

Thank you!

andyp1per, 2023-09-13:
@daniel-yavorovich do you have a version of the VTX that supports LO Race? Looks like it is by special request only.

daniel-yavorovich, 2023-09-14:
sure, my VTX works fine with Betaflight and LO Race.

dannysilence, 2023-09-20:
@daniel-yavorovich does your copter works fone in another channel/band setup?

I just was trying the same things few weeks ago having one vtx with SmartAudio and another one with Tramp and none succeed.

SerJG, 2025-01-15:
@daniel-yavorovich Hello, do you have any updates on it? It looks like I'm facing the same issue on Rush Max Solo 2.5W - L band is not working when I set it in MissionPlanner `VTX_BAND = 5` and `VTX_CHANNEL=0`, no video but everything is fine on other bands

SerJG, 2025-01-16:
> [@daniel-yavorovich](https://github.com/daniel-yavorovich) do you have a version of the VTX that supports LO Race? Looks like it is by special request only.

The VTX is [MAX SOLO VTX 2.5W](https://rushfpv.net/products/rushfpv-max-solo-vtx) in my case.

I have VTX power on the switch and tried to set the L-band from the parameters list. 
Everything works great for the R band, but for the L band, VTX switches the channel based on LED blimming, but the power level is minimal(green). It also stops reacting on the power level RC switch as well, even when I turn it all the way up. 
Even if afterward I press the power button manually - there is no any signal on the selected L channel. 
It seems that VTX goes into some weird state and is not performing properly.

The other thing I have is VTX_OPTIONS=8 (Unlocked, Pitmode is disabled), but if I turn the VTX Power switch down - it goes into pit mode, not 25 mW.

SerJG, 2025-01-16:
I noticed that L band channels are from the highest Frequency to the lowest in [code](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_VideoTX.cpp)

` { 5621, 5584, 5547, 5510, 5473, 5436, 5399, 5362}, /* LO Race */`

But the other bands range from the lowest to the highest. 

Could it affect the behavior of VTX?  

I'm unfamiliar with SmartAudio protocol implementation, but this seems odd o me.

aLDime, 2025-01-20:
SmartAudio in AP is implemented, so to speak, in strange way.. 
As I understand it, there is no interaction between manufacturers of modern VTX and AP. For example, all versions of transmitters from AKK from 3W to 16W don't work correctly! It's impossible to select the power normally, and maximum power doesn't work at all!

It's impossible to solve this issue at user level, even if you know C++ very well, since you need to know the addresses of VTX table, which are known only to manufacturer. A vicious circle.

aLDime, 2025-01-20:
> I'm unfamiliar with SmartAudio protocol implementation, but this seems odd o me.

SmartAudio in AP is implemented, so to speak, in strange way.. 
As I understand it, there is no interaction between manufacturers of modern VTX and AP. For example, all versions of transmitters from AKK from 3W to 16W don't work correctly! It's impossible to select the power normally, and maximum power doesn't work at all!

It's impossible to solve this issue at user level, even if you know C++ very well, since you need to know the addresses of VTX table, which are known only to manufacturer. A vicious circle.

andyp1per, 2025-01-23:
AP attempts to solve this without introducing custom frequency tables which would incur a large flash cost. Unfortunately manufacturers VTX's are all different and do not comply with the various specs in lots of different incompatible ways. I have spent some time trying to make this acceptable for many VTX's but I really don't use analog much anymore and no-one wants to pay me for the work so I have little incentive to fix this.

aLDime, 2025-01-25:
> no-one wants to pay me for the work so I have little incentive to fix this.

Grateful for answer, that’s basically how I understood it. I'll try to resolve this issue with AKK, at least from their side.

franky909, 2025-02-22:
> AP attempts to solve this without introducing custom frequency tables which would incur a large flash cost. Unfortunately manufacturers VTX's are all different and do not comply with the various [...]

Hello, I’m Franky from AKK, we’d like to explore how we can resolve this issue, please contact me at franky@akktek.com if you are open to discuss this further. Thank you.

andyp1per, 2025-02-22:
Hey @franky909 happy to explore. Do you have some VTX's you would like to work here? Are you able to send them to me?

franky909, 2025-02-25:
Yes, we can send our products to you.
Best Regards,
Franky
AKK TEAM
------------------------------------------------------------------

Send Time:2025 Feb. 22 (Sat.) 18:14

franky909, 2025-02-25:
> Hey [@franky909](https://github.com/franky909) happy to explore. Do you have some VTX's you would like to work here? Are you able to send them to me?

Yes, we can send our products to you. please contact me at franky@akktek.com

PR #30021 (2025-05-24): (BUG Fix) AP_VideoTX: fix VTX hang when setting LowRace band (Analog 5.8 GHz)
Fixes #24965

## Bug fix descriptin:
*Fixes an issue where setting the VTX_BAND parameter to LowRace would cause the video transmitter to hang or stop responding. Now LowRace band can be selected and used reliably without causing any malfunction.*

## Bug description:
To verify, I corrected the frequencies for the L band channels in the code and compiled a custom firmware build for the **SpeedyBee F405 V4** flight controller and a **RUSHFPV MAX SOLO VTX 2.5W** (48-channel version). After flashing this build, the issu [... cut]
closed 2025-05-24 by andyp1per