ArduPilot/ardupilot #24968: Помогите решить вопрос с дополнительной функцией в Qplane.

Refick79, 2023-09-13:
Привет.  Я собрал Q Plane, он имеет стандартную схему с v - образным хвостовыми плоскостями и управояющими поверхностями. Так же есть раскладная рама по схеме "H" тип - 1. Стандартный квадрокоптер. 
Необходимо добавить возможность настройки дополнительного релейного выхода контроллера для управления  приводом рамы в переходных режимах Plane. > Qplane и обратно. То есть по аналогии уборки шасси в  режиме Plane.  Любой переход в режим Qplane должен начинаться с разрешающего сигнала RELAY On n  TIME DALAY (X) ;  N - номер реле. X- время задержки в секундах.  Время задержки после которой произойдёт включение моторов QPlane. Так же в обратном  режиме. После отключения моторов задержка для остановки моторов и отключение релецногг пина. 
RELAY Off n TIME DALAY (X); 
Возможно данная функция уже реализована в данном проекте , не мне не удалось её обнаружить. 
Спасибо. 

**Platform*
[  Q] Plane
Ver. 4.2.
Полетный контроллер H743 matek

lixiaowei123, 2023-09-14:
Directly closed, unable to understand

IamPete1, 2023-09-14:
From translate:

Hello. I built a Q Plane, it has a standard layout with v-shaped tail planes and control surfaces. There is also a folding frame according to the “H” scheme, type - 1. Standard quadcopter.
It is necessary to add the ability to configure an additional relay output of the controller to control the frame drive in Plane transient modes. > Qplane and back. That is, by analogy with retracting the landing gear in Plane mode. Any transition to Qplane mode must begin with an enable signal RELAY On n TIME DALAY (X) ; N - relay number. X is the delay time in seconds. The delay time after which the QPlane motors will turn on. Same in reverse mode. After turning off the motors, there is a delay to stop the motors and turn off the remote pin.

RELAY Off n TIME DALAY (X);
Perhaps this function has already been implemented in this project, but I was not able to find it.
Thank you.

@Refick79 The first part of this is quite easy, you can use scripting to look for the Q mode and set the relay. The second bit is much harder, that is delaying mode switch. This is especially tricky in Qassist with the motors may come on unexpectedly. What sort of time delay would you need? Again you can use scripting to make sure the motors don't spin up until the frame is locked, but that is not ideal.

Refick79, 2023-09-14:
Задержка при отключенни равно 2 секундам.  При включении так же.  Моторы в сложенном положении рамы не включается так как на ESC сигнал не поступает по причине механического отключения от выхода контроллера. После раскрытия рамы связь возобноаляется автоматически. Поэтому меня заботит вопрос задержки запуска моторов.  Сейчас сценарий следующий : QPlane летает как самолёт,  потом включается режим посадки на esc поступает сигнал но, положение самолёта в пространстве не изменяется должным образом. Потом открывается рама, на esc подаётся сигнал шим равный 100%  акселерометра  и все  esc уходят в калибровку  .

muramura, 2023-09-14:
The official language of ardupilot's assignment and PR is English.
Do not use the National Language.

IamPete1, 2023-09-14:
@Refick79 Feel free to use whatever language your most comfortable with. There may not be a developer that can understand directly, but we will do our best with translation tools.

You can use scripting with `set_output_pwm_chan_timeout` override to ensure the outputs are 0 throttle PWM when they first connect. You can then wait some short time before removing the override and letting them function as normal. The core of your lua script might look something like this:

```
-- should fold motors if not using them currently
local motors_shutdown = motors:get_spool_state() == 0
local fold_motors = motors_shutdown and not quadplane:in_assisted_flight() and not quadplane:in_vtol_mode()

if fold_motors then
    -- Make sure the servo output channels 1, 2, 3,  and 4 stay at 1000 PWM for the next 2 seconds
    SRV_Channels:set_output_pwm_chan_timeout(0, 1000, 2000)
    SRV_Channels:set_output_pwm_chan_timeout(1, 1000, 2000)
    SRV_Channels:set_output_pwm_chan_timeout(2, 1000, 2000)
    SRV_Channels:set_output_pwm_chan_timeout(3, 1000, 2000)

end

```

You will need quite a bit more logic around that so it can run continuously and do the right thing when disarmed for example.

Its still not ideal because the vehicle is not aware that is can't use the motors, so it will still try even though there not outputting. So you will still have issues with I term wind up. Fixing that is quite complex, in the mean time a script should get you 90% of what you need.

Refick79, 2023-09-14:
Спасибо. Мой дрон отлично летает в ручном режиме но, мне важно научить его выполнять миссии. (
Я попытаюсь это реализовать через реле затвора камеры или управление шасси. 
Спасибо.