ArduPilot/ardupilot #24969: Build error:  no match for ‘operator=’ within AP_Mount_Siyi::FirmwareVersion

xianglunkai, 2023-09-13:
In current master branch.

1. build error message:
../../libraries/AP_Mount/AP_Mount_Siyi.cpp: In member function ‘void AP_Mount_Siyi::process_packet()’:
../../libraries/AP_Mount/AP_Mount_Siyi.cpp:343:9: error: no match for ‘operator=’ (operand types are ‘AP_Mount_Siyi::FirmwareVersion’ and ‘<brace-enclosed initializer list>’)
         };
         ^
compilation terminated due to -Wfatal-errors.

2.  It means bug cuased from source code like:
      // consume and display camera firmware version
      _fw_version = {
          .camera = {
              .major = _msg_buff[_msg_buff_data_start+2],  // firmware major version
              .minor = _msg_buff[_msg_buff_data_start+1],  // firmware minor version
              .patch = _msg_buff[_msg_buff_data_start+0],  // firmware revision (aka patch)
          },
          .gimbal = {
              .major = _msg_buff[_msg_buff_data_start+6],  // firmware major version
              .minor = _msg_buff[_msg_buff_data_start+5],  // firmware minor version
              .patch = _msg_buff[_msg_buff_data_start+4],  // firmware revision (aka patch)
          },
          .received = true,
      };

rmackay9, 2023-09-13:
@xianglunkai,

Thanks for the report.  I'm not seeing this issue, I wonder which compiler you're using?

xianglunkai, 2023-09-14:
@rmackay9 
Txs!
ubuntu18.04 
g++ 7.5.0

peterbarker, 2023-09-14:
On Wed, 13 Sep 2023, xianglunkai wrote:

> @rmackay9
> Txs!
> ubuntu18.04
> g++ 7.5.0

There's your problem :-)  Please use our release compiler - you can 
download it from firmware.ardupilot.org (or get it directly from Arm or 
whatever.... )

If you're compiling in SITL - you ought to be able to install a much more 
recent compiler using `apt`

Important to know we've broken older Ubuntu's in this way....

rmackay9, 2023-09-14:
Thanks @peterbarker.

@xianglunkai I will go ahead and close this issue if that's OK.  We will still see replies though.

closed 2023-09-14 by rmackay9

rmackay9, 2023-09-16:
@xianglunkai maybe this PR has resolved the problem? https://github.com/ArduPilot/ardupilot/pull/24995/files