ArduPilot/ardupilot #24974: IE800W not showing tank level

b1acksheep1, 2023-09-13:
https://github.com/ArduPilot/ardupilot/blob/ee316f04ed847e17c2f7f784cfb2c97dc0a0d778/libraries/AP_Generator/AP_Generator_IE_650_800.cpp#L43

I set up the Generator (IE 800 W) on Arducopter 4.4 with a Quadframe.
I get percentage of battery, the status and error codes BUT the tank level always shows -1.

I checked the connection and also checked if the fuel cell is sending the correct values. --> It does. (Checked via serial monitor)
The Messages look like this:
<22,71,1,0x00000000>

The percentage of the tank (first value) seems to be an integer. Why is it Multiplied by 0.01?

rmackay9, 2023-09-13:
@peterbarker, maybe this is one you can answer?  It looks like this PR is related https://github.com/ArduPilot/ardupilot/pull/23168

IamPete1, 2023-09-14:
Where are you seeing the tank level? Separate battery monitors must be setup for the fuel and battery level from the fuel cell. 

Do you have a log of it not working? (or at least a param set)

b1acksheep1, 2023-09-14:
I expect to see the tank level in Mission Planner when connected via USB. I can see the Bat1 in the Data Panel (HUD) that shows the percentage (eg. 71) and below that i see the Bat2 that is always showing the -1.
I can also see the Battery Percentage on my Herelink via QGroundControl but not the Tank Level.

Here is my first param set: 
https://pastebin.com/qsTtRU8k

I can provide a log file later that evening or tomorrow.

Thank you for your efforts in advance!

IamPete1, 2023-09-15:
@b1acksheep1 I think I have a fix, I can do a test build for you. Which flight controller are you using?

I should also warn that I was planning to remove support for the 800W model soon, my understanding was that there were not many left in the wild. I think the plan is to move to a single protocol for all units, so if/when that happens it might just be a matter of updating the firmware on the fuel cell.

b1acksheep1, 2023-09-20:
I am currently using the Holybro Pixhawk 4  (https://docs.px4.io/main/en/flight_controller/pixhawk4.html)
Will the 800 W FC still work on that single protocoll?
I guess i wont update any further.

If you could provide a test build, that would be awesome.
Thank you

IamPete1, 2023-09-20:
@b1acksheep1 Test build here: [Pixhawk4 Copter 4-4 + IE fix.zip](https://github.com/ArduPilot/ardupilot/files/12674216/Pixhawk4.Copter.4-4.%2B.IE.fix.zip)

Just copter 4.4 with the fix applied. 

To be clear on removing the protocol, nothing is decided yet, hopefully IE will provide a updated firmware so you can use the new protocol.

b1acksheep1, 2023-09-22:
Yes, now it shows the tank level and the readings are plausible.
Thank you again.

PR #24994 (2023-09-26): AP_Generator: IE 650/800: does have fuel remaining
This backend does have fuel remaining. Closes #24974
closed 2023-09-22 by b1acksheep1