ArduPilot/ardupilot #24975: Faile to compile for Obal board

akhodeir, 2023-09-13:
## Bug report

When I try to build the code, it give me the below errors

**Issue details**

```
#make obal                                                       
Setting top to                           : /ardupilot 
Setting out to                           : /ardupilot/build 
Autoconfiguration                        : enabled 
Checking for program 'python'            : /usr/bin/python 
Checking for python version >= 3.6.9     : 3.10.12 
Setting board to                         : obal 
Using toolchain                          : arm-linux-gnueabihf 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/arm-linux-gnueabihf-g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/arm-linux-gnueabihf-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 11.4.0 
Checking for need to link with librt     : not necessary 
Checking for 'lttng-ust':                : disabled 
Checking for library dl                  : yes 
Checking for program 'arm-linux-gnueabihf-pkg-config' : /usr/bin/arm-linux-gnueabihf-pkg-config 
Checking for 'libiio'                                 : not found 
Enabled OpenDroneID                                   : no 
Enabled firmware ID checking                          : no 
GPS Debug Logging                                     : no 
Enabled custom controller                             : no 
Checking for HAVE_CMATH_ISFINITE                      : yes 
Checking for HAVE_CMATH_ISINF                         : yes 
Checking for HAVE_CMATH_ISNAN                         : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE        : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE           : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE           : yes 
Checking for header endian.h                          : yes 
Checking for header byteswap.h                        : yes 
Checking for HAVE_MEMRCHR                             : yes 
Traceback (most recent call last):
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 251, in run_commands
    ctx = run_command(cmd_name)
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "/ardupilot/modules/waf/waflib/Configure.py", line 159, in execute
    super(ConfigurationContext, self).execute()
  File "/ardupilot/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/ardupilot/wscript", line 496, in configure
    cfg.get_board().configure(cfg)
  File "/ardupilot/Tools/ardupilotwaf/boards.py", line 175, in configure
    c_cpp_properties = json.load(open(os.path.join(env.SRCROOT, '.vscode/c_cpp_properties.json')))
  File "/usr/lib/python3.10/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.10/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 8 column 13 (char 159)
```

_Please describe the problem_

I created a docker image as stated on Ardupilot website and I tried to build the code using "make obal".
The python scripts fail.

**Version**
I used the latest build on mainline

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad
**Hardware type**
Raspberry Pi 4

akhodeir, 2023-09-18:
I understood what is the problem, 

My VSCode generated c_cpp_properties.json file which caused a lot of errors. 

this file is not part of the initial repo.

peterbarker, 2024-07-31:
@akhodeir glad you got this sorted out.  Closing this issue now.

closed 2024-07-31 by peterbarker