ArduPilot/ardupilot #24985: ardupilot  loiter mode new brake

lixiaowei123, 2023-09-14:
Can the loiter braking mode of APM be used to achieve continuous braking? When the expected speed is less than a certain value, the stopping point is calculated to stop the drone from moving, so that the braking distance is greater and the effect is more in line with reality

lixiaowei123, 2023-09-14:
Take a look at DJI's brakes. It's called a cool, APM's is a bit bad

IamPete1, 2023-09-14:
What do you mean by continuous braking? There is a separate parameter for the breaking acceleration in loiter, increasing it will result in more aggressive breaking. Not may of us are familiar with how DJI's fly, maybe you can explain in more detail?

rishabsingh3003, 2023-09-14:
Videos and or logs of what you are trying to achieve would be helpful to understand what you are after

lixiaowei123, 2023-09-15:
At the beginning, the braking force increases. When the braking force reaches the maximum braking force, maintain the continuous maximum braking force. When the expected speed is less than a certain value, start calculating the stop point, and then perform position control attitude

lixiaowei123, 2023-09-15:
https://www.douyin.com/video/7037432593633070350

IamPete1, 2023-09-15:
We should be able to do breaking like the video, but you will have to increase the acceleration limits.

lixiaowei123, 2023-09-18:
> We should be able to do breaking like the video, but you will have to increase the acceleration limits.

How to rotate and calculate the stop point is crucial, and speed loop control is crucial. This affects the braking posture, otherwise there will be signs of nodding back and forth to brake

lthall, 2023-09-18:
> How to rotate and calculate the stop point is crucial, and speed loop control is crucial. This affects the braking posture, otherwise there will be signs of nodding back and forth to brake

This is correct. If you have a poor tune, either the attitude controller or position controller then  you may see nodding. 

However, Loiter does not calculate a "stop point" so that part is wrong. Loiter simply slows down the velocity until it reaches zero, it doesn't care where it stops.

>> At the beginning, the braking force increases. When the braking force reaches the maximum braking force, maintain the continuous maximum braking force. When the expected speed is less than a certain value, start calculating the stop point, and then perform position control attitude

This is a very poor way of doing this.

Loiter is intended to resemble Alt_Hold when there is no wing with the additional ability to add some extra breaking when you let go of the sticks. I do intend to add a new control mode that is more robotic in feel that may be closer to what you want.

This issue doesn't describe a problem or articulate any enhancement request. I think this is more of a tuning and setup problem that should be discussed on the forums. I will close this and open it again if needed.

closed 2023-09-18 by lthall

lixiaowei123, 2023-09-18:
Thank you. Where can I see the new mode you mentioned

lixiaowei123, 2023-09-18:
> This is correct. If you have a poor tune, either the attitude controller or position controller then you may see nodding.
>
> However, Loiter does not calculate a "stop point" so that part is wrong. [...]

Thank you. Where can I see the new mode you mentioned

lthall, 2023-09-18:
> Thank you. Where can I see the new mode you mentioned

**I do intend to add a new control mode** that is more robotic in feel that may be closer to what you want.

I have not written and added the mode yet.