ArduPilot/ardupilot #24998: Webots-python SITL. Mavlink connection issue

sashacomn, 2023-09-16:
## Bug report

**Issue details**

When I'm running sitl with webots-python model I'm getting issue with mavlink connection. It says "MAV>link 1 down" 
This is command which I've used to start SITL:
`Tools/autotest/sim_vehicle.py -v ArduCopter -w --model webots-python --add-param-file=libraries/SITL/examples/Webots_Python/params/iris.parm --sim-address=172.17.208.1`

Issue happens only when use `--model webots-python`. Webots on C works fine, another models also works perfect, only `webots-python` have mavlink issue on my side

**Version**
Copter 4.4

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
SITL

**Logs**
This is console logs when sitl starting
```
Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (1m51.940s)
SIM_VEHICLE: Using defaults from (Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Adding parameters from (libraries/SITL/examples/Webots_Python/params/iris.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/alex/drone/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/alex/drone/ardupilot/build/sitl/bin/arducopter" "-S" "-w" "--model" "webots-python" "--speedup" "1" "--slave" "0" "--defaults" "Tools/autotest/default_params/copter.parm,libraries/SITL/examples/Webots_Python/params/iris.parm" "--sim-address=172.17.208.1" "-I0"
RiTW: Starting ArduCopter : /home/alex/drone/ardupilot/build/sitl/bin/arducopter -S -w --model webots-python --speedup 1 --slave 0 --defaults Tools/autotest/default_params/copter.parm,libraries/SITL/examples/Webots_Python/params/iris.parm --sim-address=172.17.208.1 -I0
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "172.17.208.1:14550" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501"
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> link 1 down
```
Logs from SITL:
```
Setting SIM_SPEEDUP=1.000000
Starting SITL Webots
Bind 127.0.0.1:9003 for SITL in
Setting Webots interface to 172.17.208.1:9002 
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Connection on serial port 5760
Loaded defaults from Tools/autotest/default_params/copter.parm,libraries/SITL/examples/Webots_Python/params/iris.parm
Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
```

In Webots 2023a I have opened `iris.wbt` . Then runned sitl. Console in webots shows this:
```
INFO: ardupilot_vehicle_controller: Starting controller: python.exe -u ardupilot_vehicle_controller.py --motors "m1_motor, m2_motor, m3_motor, m4_motor"
Listening for ardupilot SITL (I0) at 127.0.0.1:9002
Connected to ardupilot SITL (I0)
```

sashacomn, 2023-09-17:
UPDATE

Looks like there are some issue with sockets. In SIM_Webots_Python.cpp there are while loop in recv_fdm method. So this loop is working until it will get some reply from Webots app. But for some reasons webots send messages but this messages is not reaching sitl in port (9003)

IanBurwell, 2023-11-27:
Sorry, I only just saw this. What OS are you running SITL on, and are you trying to use WSL if you are using windows?

sashacomn, 2023-11-27:
> Sorry, I only just saw this. What OS are you running SITL on, and are you trying to use WSL if you are using windows?

Yeah, I’ve used Windows 10 and WSL 2.

IanBurwell, 2023-11-29:
@sashacomn I was able to reproduce what you are seeing by commenting out the code in the webots controller that sends data to the SITL. It would seem that data is correctly being sent from the SITL to webots (as webots detects the SITL), but not the other way around. I think that either your windows firewall is blocking this communication, or you need to add your WSL2 IP to `iris.wbt` with a text editor. Copying from the wiki:
```
You may also need to add WSL’s IP found with "hostname -I" to iris.wbt so it looks something like this:

Iris {
   controller "ardupilot_vehicle_controller"
   controllerArgs [
      "--motors"
      "m1_motor, m2_motor, m3_motor, m4_motor"
      "--sitl-address"
      "172.x.x.x"
   ]
}
```

devnim, 2023-12-27:
@IanBurwell. 

I also have an issue where SITL cannot connect to Webots. I have Windows 11 and using WSL2. I followed the tutorial for the first-time setup and I added sitl address to iris.wbt param file as per instructions. And I get a MAV> link 1 down. 

I am actually wandering whether the underlying issue is related to another existing issue with Airsim. Open issue #25257 (https://github.com/ArduPilot/ardupilot/issues/25257). That issue is recent and also shared by other issuers and discussed in the Ardupilot forum. This is a place that Ardupilot tutorial pointed to to get WSL2 and Airsim on Windows to talk. See the most recent posts on the bottom of https://discuss.ardupilot.org/t/gsoc-2019-airsim-simulator-support-for-ardupilot-sitl-part-ii/46395/23). 

Before I tried to get Webots - SITL to work, I was trying to get Airsim (Colosseum) and SITL to connect and could not. I followed the instructions carefully. I am able to run SITL and Airsim separately without issues, so it seems to me the issue is with the connection. Since I am having an issue with Webots - SITL connection as well, I wander whether it is an issue between WLS and Windows, perhaps caused by a recent Windows update. I am only guessing. You mentioned potential blockage by firewall and I wander if that could be the cause?

sairaghavamunagala, 2024-04-29:
1. Run the following command in the terminal
`ardupilot/Tools/autotest/sim_vehicle.py -v ArduCopter -w --model webots-python --add-param-file=ardupilot/libraries/SITL/examples/Webots_Python/params/iris.parm `
2. Open the webots and open the iris world.
3. It will show a message connected.

Jadeiin, 2024-10-23:
I had encountered the same problem, generally this can be related to mavproxy, after changing the IP address of the listener represented by the “--out” parameter I managed to fix the problem.