ArduPilot/ardupilot #25007: direct modification of GPIO data registers may be vulnerable to read-modify-write races

rsaxvc, 2023-09-17:
I have not seen this issue on ArduPilot on my Raspberry Pi, but I have seen this on several other RISC devices, and suspect I'm getting lucky. JamesPattison asked me to create this ticket: https://discord.com/channels/674039678562861068/854213651250020372/1152720064395616286

## Bug report

**Issue details**

Some GPIO direct hardware register accesses appear to have data races between both ArduPilot threads as well as other userspace and kernel threads. How often this may occur will vary widely with selected board.

**Version**
master / d9818ae1abfef10caf2 , nearly all releases supporting BBB and RPI.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Basic peripheral driver issue unrelated to airframe.

**Hardware type**
On each GPIO write(): PXF, ErleBoard, BBBMini, Blue, Pocket
On each GPIO pinMode(): RPI-based boards

**Logs**

I pulled the following primitives out into godbolt(ARM GCC 11.2.1 -O3 -mcpu=cortex-a9 -ffunction-sections) so it may not exactly match instruction for instruction in released firmware, but the data races are present in both RPI and BBB builds.
```
Linux::GPIO_BBB::pinMode(unsigned char, unsigned char):
        tst     r1, #128
        bxne    lr
        lsr     r3, r1, #5
        cmp     r2, #0
        and     r1, r1, #31
        add     r0, r0, r3, lsl #4
        mov     ip, #1
        ldr     r2, [r0, #4]
        ldr     r3, [r2] //Data race begin
        orreq   r3, r3, ip, lsl r1
        bicne   r3, r3, ip, lsl r1
        str     r3, [r2]//Data race end
        bx      lr

Linux::GPIO_BBB::write(unsigned char, unsigned char):
        tst     r1, #128
        bxne    lr
        lsr     r3, r1, #5
        cmp     r2, #0
        and     r1, r1, #31
        add     r0, r0, r3, lsl #4
        mov     ip, #1
        ldr     r2, [r0, #12]
        ldr     r3, [r2]//data race begin
        biceq   r3, r3, ip, lsl r1
        orrne   r3, r3, ip, lsl r1
        str     r3, [r2]//data race end
        bx      lr
```

Compared to the Raspberry Pi implementation, whose CPU supports atomic write-one-to-set and write-one-to-clear registers, but only for the data registers(not the mode register):

```
Linux::GPIO_RPI::pinMode(unsigned char, unsigned char):
        cmp     r2, #0
        bne     .L11
        movw    r3, #52429
        ldr     ip, [r0]
        movt    r3, 52428
        umull   r2, r3, r3, r1
        lsr     r3, r3, #3
        add     r2, r3, r3, lsl #2
        ldr     r0, [ip, r3, lsl #2]//begin data race
        sub     r2, r1, r2, lsl #1
        mov     r1, #7
        add     r2, r2, r2, lsl #1
        uxtb    r2, r2
        bic     r2, r0, r1, lsl r2
        str     r2, [ip, r3, lsl #2]//end data race
        bx      lr
.L11:
        movw    r2, #52429
        ldr     ip, [r0]
        movt    r2, 52428
        mov     r3, #7
        umull   r0, r2, r2, r1
        lsr     r2, r2, #3
        add     r0, r2, r2, lsl #2
        sub     r1, r1, r0, lsl #1
        ldr     r0, [ip, r2, lsl #2]
        add     r1, r1, r1, lsl #1
        uxtb    r1, r1
        mvn     r3, r3, lsl r1
        and     r0, r0, r3
        str     r0, [ip, r2, lsl #2]
        ldr     r0, [ip, r2, lsl #2]//begin data race
        and     r3, r3, r0
        mov     r0, #1
        orr     r3, r3, r0, lsl r1
        str     r3, [ip, r2, lsl #2] //end data race
        bx      lr

//no data race. Constant is constructed in r3 with bits corresponding to GPIO pins to be cleared, then atomically written to gpclr0 or gpset0.
Linux::GPIO_RPI::write(unsigned char, unsigned char):
        cmp     r2, #0
        mov     r3, #1
        ldr     r2, [r0]
        lsl     r3, r3, r1
        strne   r3, [r2, #28]
        streq   r3, [r2, #40]
        bx      lr
```

**Problem Analysis**
If each thread conducts LDR, OR/BIC/AND, STR, it's likely that rare cases where the following operation order may occur(interruptions and context switches not shown): Thread1LDR, Thread2LDR, Thread2OR/BIC/AND, Thread2STR, Thread1OR/BIC/AND, Thread1STR. This could occur on parallel threads in an SMP device, or occur due to a context switch during the instructions I've annotated as 'data race' on even a single core device. Note that these register accesses aren't limited to ardupilot - other userspace thread using mmap() or sysfs()/GPIO or libgpiod or kernel threads touching the kernel GPIO interfaces at the same time could also trigger this. This is likely to be quite rare in practice, but I've seen it fail on other ARM systems.

**Some Potential Solutions**

1. Could move BBB and RPI based boards to sysfs like the other boards, which will go through the kernel GPIO interfaces, whose backends are usually protected through a spinlock. There will be some overhead to do so.
2. Could move BBB and RPI operations that cannot be synchronized to sysfs, while keeping others directly. For BBB, this would mean pinMode() and write() go through sysfs, and read() would remain as register-mapped. For RPI, only pinMode() would go through sysfs, and write() and read() would remain as register-mapped.
3. Could hope it doesn't happen often enough to justify preventing.

**Notes**

- I've mentioned sysfs quite a bit, whose GPIO interface is deprecated. Changing to libgpiod is considered separate from this, and could be substituted anywhere I mentioned sysfs/GPIO.
- Currently neither BBB nor RPI use a memory barrier instruction to ensure accesses to the GPIO registers are not reordered with accesses to other peripherals. This should probably be considered separately, at least for boards with multiple GPIO banks if there are peripherals wired across banks, or if relying on access ordering with other peripherals, like SPI or I2C.

rsaxvc, 2023-09-17:
Bug confirmed on Raspberry Pi with the following tests(See https://github.com/rsaxvc/ardupilot/pull/new/linux_gpio_racer ):

- New ArduPilot unit test that toggles pin 12 from input to output, verifying afterward that the change worked.
- Bash loop hammering raspi-gpio set ${UNRELATED_PIN} ip;raspi-gpio set ${UNRELATED_PIN} op

ArduPilot test tends to die pretty quickly once both programs are running. But sometimes it runs a long time. And sometimes it dies immediately. My board only configures GPIO direction once during startup, so once I'm up, I'm up.

rsaxvc, 2023-10-01:
FYI: @lucasdemarchi , @Williangalvani . Could we add Linux tag?

lucasdemarchi, 2023-10-01:
It's an expected race when accessing the HW directly bypassing the kernel. There's no possible synchronization in user space if you account for other programs.

For ardupilot's threads it would be a bug indeed.

I once had a GitHub issue to convert all the boards (that can be converted) to sysfs. However note that that interface is much slower.  If we are moving to the proper kernel abstraction, then we should go directly to the gpiod interface through the ioctl.

rsaxvc, 2023-10-01:
If I understand it right, as long as AP never shares individual pins with other programs/kernel:

Pin configuration through direct access: almost never safe.

GPIO read through direct register read: always safe.

GPIO write through direct register write: requires GPIO SET/CLEAR registers(not hooked up on AP BBB yet).

So I think we could do a hybrid implement. Pin configuration always through sysfs or gpiod, and low-overhead GPIO read & writes using direct hardware registers when it is safe to do so, otherwise falling back to gpiod/sysfs.