ArduPilot/ardupilot #25022: APM: Add support webserver？

xianglunkai, 2023-09-19:
I have been working on embedded web recently, and I feel that it has great benefits for the application of embedded devices. I wonder if Ardupilot has considered this aspect?
If there is a need for this, I can try contributing some code.

rmackay9, 2023-09-19:
@xianglunkai,

I guess you mean a small webserver running on the autopilot?  If "yes" then I think the closest we have is [APWeb](https://github.com/ArduPilot/APWeb) which was included with the SkyViper product.

For companion computers, I think RPanion and BlueRobotics BlueOS are perhaps the most advanced solutions.

rmackay9, 2023-09-19:
Somewhat related Peter Hall has been adding a lot of new tools to the AP web server (on the AP servers not on autopilots) https://github.com/ArduPilot/WebTools.

I might close this issue because it's really a question.  Of course I'll see your replies though.  Thanks!

closed 2023-09-19 by rmackay9