ArduPilot/ardupilot #25025: ICM42688 "stuck" gyro values

tridge, 2023-09-19:
This is to track an issue seen on 3 vehicles where the ICM42688 can get "stuck" gyro data.
First seen on a CubeOrangePlus copter first IMU Y axis in this log:
[image]
next seen on an OctoQuad copter, also with CubeOrangePlus, this time on X axis:
[image]
this was with raw logging, and the raw values show the repeated data is bitwise identical:
[image]
next seen on a MatekH743-bdshot first IMU (also an ICM42688), with batch logging. After unwinding the batch logging we see:
[image]
[image]

tridge, 2023-09-25:
changed ODR to 32kHz and see 45 identical samples in a row:
[image]

tridge, 2023-09-25:
the timestamp in the FIFO packet doesn't do anything strange when we get the stuck values:
[image]
but it is interesting that at 8kHz we don't get 125 difference in timestamp values:
[image]

vgade-invn, 2023-10-21:
The problem is most likely caused by AFSR mode of the sensor. 

AFSR is a sensor setting used to fine-tune the device for lower noise density at lower measurement ranges, while still providing the full sensor range. At around 100 dps, you can observe the sensor switching from low-range (low noise) to high-range (higher noise) mode. When you disable the AFSR, the device will always be in high-range mode, with a slight reduction in noise density at FSR < 100 dps.

To disable AFSR, set bit [7:6] = 01 at the address of 77 at bank 0 register. The default bit [7:6] = 10 which enables AFSR.

Thanks,
Vishal Gade,
TDK Invensense.

tridge, 2023-10-21:
@vgade-invn many thanks!
does this only apply to the ICM42688 or does this affect some other parts as well?

vgade-invn, 2023-10-21:
This applies to all ICM-42xxx and IIM-42xxx parts.

Thanks,
Vishal.

tridge, 2023-10-22:
> This applies to all ICM-42xxx and IIM-42xxx parts.

@vgade-invn we have not seen the issue on other than the ICM42688. Could it be that this setting applies to all the IxM-42xxx, but the actual "stuck" data issue is ICM42688 specific?

closed 2023-10-22 by tridge