ArduPilot/ardupilot #25034: Plane refused to climb in any autothrottle mode.

AndKe, 2023-09-19:
## Bug report

**Issue details**
Why flying in guided mode, we went to about FL80 when we discovered that we were unable to command a further climb (we went to FL100 several times before)

No altitude command that required climb, resulted in a climb, we could command a lower altitude, but not a higher (the plane would not try to climb -just cruising level).

We tried to issue the same command from another GCS.
We uploaded a mission , but AUTO would not climb.
Descended and switched into Cruise : unable to climb
then FBWA  - seems to climb
then  FBWB - unable to climb
then stabilize, which of course could climb...
Qassist started in a turn, switched momentarily to QLoiter and Cruise - and then, suddenly, the plane would climb.
From now on, it could climb in Guided mode again!

This seems to be some (potentially dangerous) bug in the altitude-controlled flight modes.
For a far/distant flight where one need to climb to return home, it could be catastrophic.

**Version**
4.4.0 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
CubeOrange

**Logs**
https://drive.google.com/drive/folders/1iF0CmcsuafuWiayj3e61wqbcxVL72bDz?usp=sharing

magicrub, 2023-09-19:
Thanks for reporting and including a log. We'll take a look.

IamPete1, 2023-09-19:
Not found a clear cause, but have got some steps on the way I think. 

First of all, the height demand was getting into TECS, so this is not a problem in the plane code. We see `TECS.hin` at the requested altitude, but `TECS.hdem` stops increasing. 

[image]

Also note in `TEC2.KI` is 0 for the whole flight. KI is calculated here:

https://github.com/ArduPilot/ardupilot/blob/4a15d9cc1bedb9b95e8540216340759bf7453ec3/libraries/AP_TECS/AP_TECS.cpp#L1010-L1021

For it to be zero `inhibit_integrator` must be true. This can happen if `_pitch_dem_unc > _PITCHmaxf`. We see that same condition here:

https://github.com/ArduPilot/ardupilot/blob/4a15d9cc1bedb9b95e8540216340759bf7453ec3/libraries/AP_TECS/AP_TECS.cpp#L1187

So `max_climb_condition` would also be true. This causes the height demean to never increase. 

https://github.com/ArduPilot/ardupilot/blob/4a15d9cc1bedb9b95e8540216340759bf7453ec3/libraries/AP_TECS/AP_TECS.cpp#L1190-L1191

So, I think that we were hitting that `max_climb_condition` case. I'm not sure why that is. I'm not sure why `inhibit_integrator` is true for the whole flight but the vehicle can climb initially and then again at the end.

priseborough, 2023-09-20:
This plane was operating with ARSPD_USE =0 which TECS handles as a special casein the pitch calculation  by setting _SKE_weighting to 0. This is why the kinetic energy error integrator remains 0. I have am currently looking at a log from another user also flying a VTOL tailsitter  with ARSPED_USE = 0 where the TECS internal height demand wouldn't increase. I'll investigate further.

AndKe, 2023-09-20:
@priseborough Good catch. We disabled the ARSPD_USE during the previous flight when we suspected icing to be affecting the pitot tube.  (we had trace icing and light icing at some altitudes - observed by the leading-edge camera. )

IamPete1, 2023-09-28:
closed by https://github.com/ArduPilot/ardupilot/pull/25131

closed 2023-09-28 by IamPete1