ArduPilot/ardupilot #25041: ArduCopter 4.4 list non-exist SPI compass

mw66, 2023-09-20:
## Bug report

**Issue details**

https://discuss.ardupilot.org/t/copter-4-4-cannot-find-compass-3-while-3-6-12-has-no-problem/106423

Basically, ArduCopter 4.4 list non-exist SPI compass.

Please see the comparison between 4.4 and 3.6.12 screenshot on the above page.

**Version**
ArduCopter 4.4

**Platform**
[X ] Copter

**Hardware type**

Pixhawk 2.4.8 

https://discuss.ardupilot.org/t/copter-4-4-cannot-find-compass-3-while-3-6-12-has-no-problem/106423

mw66, 2023-09-21:
I know, but is this a software bug? why that chip (SPI, LSM303D) is listed at all?

amilcarlucas, 2023-09-22:
Probably a bug in 3.6 that no longer is present in 4.4.

rmackay9, 2023-09-22:
Hi @mw66,

We keep the older compasses around in the list so that if they re-appear the user doesn't have to re-do the calibration.  On the MP screen there is also a button to remove the old ones in case they're getting in the way for some reason.

closed 2023-09-22 by rmackay9