ArduPilot/ardupilot #25059: EKF Uncontrolled height, incorrect vertical velocity obtained from inertial navigation

lixiaowei123, 2023-09-20:
I recently encountered a problem where the altitude of the multi rotor drone is not controlled. From the logs, it can be seen that there is a problem with the vertical speed, but I don't know what caused it

lixiaowei123, 2023-09-20:
[log.zip](https://github.com/ArduPilot/ardupilot/files/12678538/log.zip)

lixiaowei123, 2023-09-20:
[image]

rmackay9, 2023-09-21:
Hi @lixiaowei123,

It looks like high vibration when the throttle is raised.
[image]

By the way, support questions should really go in the forums.  The issues list is for confirmed bugs or enhancement requests so I'll close this.  Sorry to be strict but if we're not then our to-do list just turns into a support forum.

closed 2023-09-21 by rmackay9

lixiaowei123, 2023-09-21:
@

> Hi @lixiaowei123,
>
> It looks like high vibration when the throttle is raised. [image]
>
> By the way, support questions should really go in the forums. The issues list is for confirmed bugs or [...]

thank you,I feel that there may be a bug in the acceleration offset learning of EKF
[image]