ArduPilot/ardupilot #25083: Ardupilot crashes when TECS_HDEM_TCONST=0

suraqis, 2023-09-23:
## Bug report

**Issue details**

When parameter TECS_HDEM_TCONST is set to 0 (out of range, but makes some sense) ardupilot crashes due to divide by zero.

From the description, it is clear that HDEM is filtered with filter time constant TCONST. Its zero value intuitively has to turn off filtering, so it may be wanted by the user.

**Version**
Arduplane 4.4+

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane

**Hardware type**
Any (tested on SITL)

**Logs**
Trivialy reproducible

IamPete1, 2023-09-23:
Note that a flight controller will not crash because of this. However the nan's might end-up doing odd stuff. You could take out the panic from the error and see what would happen in real life.

suraqis, 2023-09-24:
That's good, but for this case I'd prefer to have filtering turned off if it's zero