ArduPilot/ardupilot #25142: Internal error 0x1000000 cube blue H7

harelevy1, 2023-09-29:
## Bug report

After a flight of about 3 minutes I get this message  'Internal error 0x1000000'
Both in manual flight and automatic flight The flight controller is a cube blue with an AUTOPILOT board

After receiving the message, the drone enters FAIL SAFE and I cannot continue the mission, it should be noted that it continues to fly normally after receiving the error

I use a cube blue but there is a version of an cube orange on it
Version 4.1.5 copter cube orange 

**Platform**

[  ] Copter

**Airframe type**
normal copter

**Hardware type**
Pixhawk, Cube blue

[image]
[image]

IamPete1, 2023-09-29:
Does the issue happen on the current stable release, 4.4.1?

harelevy1, 2023-09-29:
I can check, the problem is that I need version 4.1.5 because all my other
drones are on this version.

Does the new version have a fix for this problem?

magicrub, 2023-09-29:
For reference, error 0x1000000 is when the IMU has had to reset itself too many times. Has this hardware had any crashes on it?

harelevy1, 2023-09-29:
Hi, no, the hardware is new, and the drone hardly flies; it has about 3 hours of flight time.
It never fall or crashed.

be-ska, 2023-10-02:
On a custom hardware we're testing the error 0x1000000 was caused by a soldering issue on one IMU. I don't have any experience with Cube blue, but maybe you could check on a disarmed log if the 3 IMUs are coherent between each others.

harelevy1, 2023-10-02:
I uploaded the logs from the flights
Where can I find what you said?

https://drive.google.com/drive/folders/1-OU3ucBDJILU8RPmWrdVw98y5yC4FWgl?usp=sharing

be-ska, 2023-10-05:
Seems like a Cube Orange issue that [has been solved since 4.3.3](https://discuss.cubepilot.org/t/cube-orange-internal-errors-1000000/10238/3)