ArduPilot/ardupilot #25186: Copter: choose RC channel for Transmitter Based Tuning

be-ska, 2023-10-05:
## Feature request
At the moment users need to assign a knob on CH6 for [Transmitter Based Tuning](https://ardupilot.org/copter/docs/common-transmitter-tuning.html). 
This could be an issue, in particular when setting up Traditional Helicopters, because usually CH6 is used for pitch control and CH3 is used for throttle. On some cheap transmitters users can't change the channels map, so there's no way to use the Transmitter Based Tuning feature, even if the transmitter have an available knob or slider.

I think there are two **solutions** for this small issues:
1. adding a parameter like `TUNE_CH` where users set the tune channel;
2. adding a `RCx_OPTION` function for the transmitter based function.

If this feature would be considered and someone would advice me to procede with solution 1 or 2, I can submit a pull request to add this small improvement.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2023-10-06:
>     2. adding a `RCx_OPTION` function for the transmitter based function.

This one; I've considered doing this previously...

> If this feature would be considered and someone would advice me to procede with solution 1 or 2, I can submit a pull request to add this small improvement.

Great!

peterbarker, 2024-07-31:
We've merged @be-ska 's PR which gives us this functionality - thanks @be-ska !