ArduPilot/ardupilot #25210: LOG_BACKEND_TYPE causes gettings params forever

stevempotter, 2023-10-08:
## Bug report

**Issue details**

I am trying to set up blackbox logging to SD card on a Matek F405-VTOL flight controller.
I tried setting LOG_BACKEND_TYPE to 1 in Mission Planner. The next time I tried to connect via USB, I could not get past the "Getting params" popup window. It kept getting stuck at about 95% of the way of the progress bar, and would say "already got parameter X" a lot.
Cancelling this window just leads to a message saying "Did not finish getting parameters. Try again?" and there is no option to NOT try again, so I can't get to the Full Parameter List to set LOG_BACKEND_TYPE back to 0. All I could think to do to get a connection to the FC again was to reflash the FC firmware with STM32 Cube, and reload all my params from a saved .params file.

There should be a way out of a failed "get params". Or at least more information about why it failed.

**Version**
4.4.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heewing T-1 Ranger VTOL (tilt-motor tricopter)

**Hardware type**
Matek F405-VTOL
SD card is SanDisk Extreme formatted as FAT32
**Logs**
CANT GET LOGGING WORKING! 
If you have suggested settings that will enable logging to SD, please put them in a reply!
Thanks!

peterbarker, 2023-10-18:
@andyp1per do you have one of these controllers?  I'm guessing you haven't seen these problems?

andyp1per, 2023-10-18:
I have lent mine to a friend - I will ask him to test.

andyp1per, 2023-10-20:
Tested and it works fine

closed 2023-10-20 by andyp1per

stevempotter, 2023-10-20:
Can you please list your logging-related settings so I can be sure to use those?
(or a whole .param file?)

stevempotter, 2023-10-20:
As I said above, There should be a way out of a failed "get params". Or at least more information about why it failed.

andyp1per, 2023-10-20:
Hey @stevempotter my suspicion is that this is RAM related. If ftp fails its probably because there was not enough RAM to start - so nothing to do with the logging. I suggest you reset to default parameters, check that it all works and then slowly add back your parameters to see what tips you over the edge.